pikvm/pikvm #186: Black and white option

sharicov, 2021-02-14:
Could you please add black and white option for better image compression?
Thanks in advance

mdevaev, 2021-02-14:
I'm working on H.264 compression, which will be much better than a black and white image. I won't be doing this right now, but I'll leave this iss for the future, as it's useful.

mdevaev, 2021-07-04:
Since [H.264](https://github.com/pikvm/pikvm/blob/master/pages/webrtc.md) is now available and testing shows a significant reduction in traffic consumption, I don't think that the black-and-white image option is still relevant. The ancient KVMs used it because they did not have differential encoding, but now we have. In addition, to get a black-and-white image from RGB, processing on the CPU is required, since hardware conversion is not available for most encoding methods. So I'm closing it as unnecessary, unless something changes completely in the future.

closed 2021-07-04 by mdevaev