pikvm/pikvm #187: Bluetooth keyboard

sharicov, 2021-02-14:
Could you please add an option to connect to Raspberry with bluetooth keyboard and use it to control the host compute.
Very useful when only have BT keyboard (with no second PC) and want to control the host's computer BIOS.
Thanks in advance.

alexsalex, 2021-02-14:
May be this helps you?

https://github.com/pikvm/pikvm/blob/master/pages/bluetooth_hid.md

sharicov, 2021-02-15:
> May be this helps you?
> 
> https://github.com/pikvm/pikvm/blob/master/pages/bluetooth_hid.md

Unfortunately not, I am looking for something quite opposite. I would like to connect keyboard (mouse) to Raspberry in order to control host computer connected to pikvm over USB. Couldn't set up bluetooth stack on pi-kvm.
Tried several guides (example: https://wiki.archlinux.org/index.php/bluetooth_keyboard) with no success.

mdevaev, 2021-02-16:
In general, you need not only Bluetooth, but also USB (first of all). I'll save it for the future.

mdevaev, 2025-05-23:
Released. To enable this feature, update OS with `pikvm-update` and run `rw; systemctl enable --now kvmd-localhid; ro`.

In this mode, all USB keyboards and mouses connected to PiKVM directly will be forwarded the host. If you're using PiKVM [PiKVM Switch](https://docs.pikvm.org/switch), they will be forwarded via the Switch to the active selected host. There are several hotkeys available that are always active:

* `LeftAlt, LeftAlt, K` - (mnemonic **K**VM, quick hit) disable keyboard/mouse grabbing and allow them to use with PiKVM locally, for example, for the console operating.
* `LeftAlt, LeftAlt, H` - (**H**ost) - switch bag to the passthrough mode and pass keyboard-mouse events to the host.
* Switching the PiKVM Switch channels. If you have one or two switches, you can use `LeftAlt, LeftAlt, 1` (1-8) to switch between 8 channels. For three or more Switches, you need to use double numbers, like `LeftAlt, LeftAlt, 3, 2` (unit 3, channel 2).

Now you can set up your workplace so that all your input devices are connected via PiKVM, this is especially convenient with the V4 Plus, which can also passthrough a monitor. And if you use a switch, then you will no longer need to have a separate keyboard and mouse for local work. Your workplace behaves exactly the same locally as it does remotely.