pikvm/pikvm #188: Cannot mount as flash

sharicov, 2021-02-14:
iso was mounted as cdrom.
When switching to flash and pressing "Connect drive to server", getting following error:
Can't configure MSD:
{"ok: false, "result": {error": MsdlsBusyError", "error_msg":
"Performing another MSD operation, please try again later"}}

Also would be extremely nice if error messages can be selected, so it would be easier to submit it rather than typing it manually.
Thanks

mdevaev, 2021-02-15:
Probably the operating mode did not have time to change and you immediately performed another operation.

sharicov, 2021-02-16:
Still struggling, using both manuals:
https://github.com/pikvm/pikvm/blob/master/pages/cookbook.md#create-a-microsoft-windows-based-flash-disk-image
https://github.com/pikvm/pikvm/blob/master/pages/cookbook.md#mass-storage-drives
kvmd-otgmsd -i 1 --set-rw=1 --set-cdrom=0 --set-image=/root/flash.img **- works**
web gui with the same image doesn't.
Web gui shows image (flash) as connected, but in the windows file explorer it shows greyed out removable storage. Disk manager shows Removable: No media.

mdevaev, 2021-02-16:
If you change the media type, you need to either reconnect the device (this is done on v3) or reboot the host.

sharicov, 2021-02-16:
Ok, thanks, but then, how does it work with: kvmd-otgmsd -i 1 --set-rw=1
--set-cdrom=0 --set-image=/root/flash.img ?
I did try reset in the GUI, didn't work either.
Is there any estimate on when v3 will be released?

mdevaev, 2021-03-01:
The web interface cannot work with the second (and so on) device, it only serves the first one in read-only mode. Contact me in discord. You seem to know russian, and it will go faster.

v3 should be available for purchase at the end of March

closed 2021-07-04 by mdevaev