pikvm/pikvm #189: Warning window with confirmation on remove image

sharicov, 2021-02-14:
Please add warning window with confirmation on image remove.
Currently it removes it immediately, very inconvenient.
Thanks

mdevaev, 2021-02-14:
This window has been there for a long time. Maybe I didn't understand you?

[image]

sharicov, 2021-02-15:
> This window has been there for a long time. Maybe I didn't understand you?
> 
> [image]

You are absolutely write, I suppose I didn't understand it correctly and pressed OK automatically. Was under impression it is removing it from the host computer, not deleting it from the pi-kvm. Maybe it worth to highlight it in the message. Something like: Are you sure ... Image will be deleted from the Pi-KVM.

mdevaev, 2021-02-15:
Okay, fixed

closed 2021-02-15 by mdevaev