pikvm/pikvm #190: rngd daemon load cpu for 100%

alexsalex, 2021-02-14:
Subj.

Cant reproduce this issue, the issue come after disable daemon and enable daemon again.

RasPi Zero W

[image]

alexsalex, 2021-02-14:
After reboot issue came back...

[image]

mdevaev, 2021-02-14:
What about `systemctl disable rngd`?

alexsalex, 2021-02-14:
what about it? It works, disable the daemon and the load go to normal... Do I need this daemon?

mdevaev, 2021-02-14:
This is random number generation and it speeds up handshake https a bit. I think you can turn it off, since there are problems with it. It wasn't there before. RedHat seems to have broken everything.

th3drk0ne, 2021-02-15:
switch rngd service to the hardware number generator instead of jitter which uses the cpu. 
All raspberry pi models have hardware number generators including the zero. 
Entropy pool will be 3000+ immediately after boot instead of slowly generating the pool using the cpu on boot
[https://blog.thewatertower.org/2019/05/13/randomness-on-raspberry-pi/]

mkdir /etc/systemd/system/rngd.service.d
cat > /etc/systemd/system/rngd.service.d/params.conf <<EOF
# disable jitter source
[Service]
ExecStart=
ExecStart=/sbin/rngd --foreground --debug --exclude 5
EOF
systemctl daemon-reload
systemctl restart rng

@mdevaev could this be added by default to the pi zero build working cleanly on my setup for weeks with a custom ssl with no performance impact

mdevaev, 2021-02-16:
In case of arch it will be enough to edit `/etc/conf.d/rngd` and append option `-x 5`.

mdevaev, 2021-02-16:
Done

closed 2021-02-16 by mdevaev

wnark, 2021-06-22:
Hello, this problem may still exist in the new version of the mirror
[image]

wnark, 2021-06-22:
I use
systemctl disable rngd
systemctl stop rngd
Come to stop rngd, don't know what will happen.

wnark, 2021-06-22:
[image]
I turned off rgnd, but it still seems to be able to generate random numbers somewhere, rngtest is normal, so pikvm can systemctl disable rngd

mdevaev, 2021-07-31:
`echo 'RNGD_OPTS="-o /dev/random -r /dev/hwrng -x jitter -x pkcs11 -x rtlsdr"' > /etc/conf.d/rngd` and reboot