pikvm/pikvm #191: Server doesn't start if USB from PiKVM inserted

alexsalex, 2021-02-14:
Kind of interested issue.

My server seated up to power on when the power connected. But if the PiKVM connected and powered from external power supply the server do not start. Take off the USB, server starts without any issue.

Will provide any info by request.

mdevaev, 2021-02-14:
This case is described in the documentation. You need to remove the power wire from USB-C.

alexsalex, 2021-02-14:
Emmm... I don't have USB-C, using micro usb only for **raspberry pi zero w**. RasPi connected to power because sometimes I need to do a power cycle on the server.

mdevaev, 2021-02-14:
Okay, in your case, you have two micro usb devices. One of them goes to the power supply, the other is connected to the computer. Through the Pi, power is supplied to the USB server, this is the reason. Take the OTG wire and cut off the power wire there. You can also seal it in the USB-A end.

mdevaev, 2021-02-18:
I think I've answered your question)

closed 2021-02-18 by mdevaev