pikvm/pikvm #193: Ethernet module with SPI interface

wnark, 2021-02-15:
Can this support Ethernet module using SPI interface? This allows the Raspberry Pi ZEROW to have an RJ45 interface
like this

[image]

the blog:
https://shumeipai.nxez.com/2017/10/10/raspberry-pi-spi-netword-configuration-tutorial.html

wnark, 2021-02-15:
Even, the Raspberry Pi can use the 4 / 5G module to access the Internet. Can the 4G-SPI module be used instead of wifi?

like this:https://blog.csdn.net/huayucong/article/details/51706307

You may need Google Translate, LOL，thanks

mdevaev, 2021-02-15:
We have a regular Raspbian kernel. If this is supported, then everything will work. In extreme cases, you will have to compile the necessary module.

wnark, 2021-02-15:
> We have a regular Raspbian kernel. If this is supported, then everything will work. In extreme cases, you will have to compile the necessary module.

I found another magic chip, ch9329, which can simulate spi as USB, which saves an arduino
I try to see if the Raspberry Pi can use it to communicate with the computer

mdevaev, 2021-02-18:
I believe I have answered your question, so this issue will be closed. Write here or in the new issue if you want to discuss something else. I also recommend discord, as we have Zero users who use SPI Ethernet.

closed 2021-02-18 by mdevaev