pikvm/pikvm #194: Wifi doesn't work after first reboot

sharicov, 2021-02-15:
Raspberry zero w
Ready image from here: https://pikvm.org/download.html (v2-hdmi-zerow.img.bz2)
Set up WiFi as described in the guide: https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md#the-final-steps.
After first boot waited 10 minutes then rebooted by plugging out usb from host computer.
Wifi didn't ever connect (not after third boot either).
Tried it twice wiping the SD card.
The only solution I found is to create pikvm.txt again, to add:
FIRSTBOOT=1
WIFI_ESSID="mynet"
WIFI_PASSWD="p@s$$w0rd"

And to raspberry run the setup again. Then, on a second boot wifi connects.
Reproduced it twice, wiping the SD in between. Worked both times.
Please fix.
Thanks

mdevaev, 2021-02-15:
I didn't understand anything. After flashing the memory card, you must edit the file pikvm.txt, then boot the OS, wait 10 minutes, and restart. After that, everything will work. Isn't that right? Describe it point by point, I'm confused.

sharicov, 2021-02-15:
Sorry making it complicated. Step by step:
1, Flashed the card
2. Edited the pikvm.txt
3. Inserted card into raspberry zero w
4. Booted for the first time
5. Waited 10 minutes
6. Restarted
7. No wifi connection
8. Removed the card
9. created pikvm.txt
10. edited just like first time (with firstbooot=1 and wifi credentials)
11. back to stage 3-6
12. wifi connected
Done twice from scratch to verify (including wipe of SD)

Thanks for your support

mdevaev, 2021-02-15:
Are you editing a file with windows or linux? I just downloaded a new image, tested it, and got wifi after the first reboot.

sharicov, 2021-02-15:
windows notepad both times

mdevaev, 2021-02-15:
I can't reproduce it. Make a new memory card, edit the file pikvm.txt and send it to me. Important: do not copy the contents, copy the entire file. If it's about hidden symbols, I need to know. You can write a fake essid and password, but if they contain spaces or other special characters, write this.

bbeaudoin, 2021-02-15:
@sharicov, it appears the instructions already match your fix. You mentioned two tried this twice and both times succeeded. I'm going to consider the issue resolved though you may want to consider setting the wireless regulatory domain via a new option called `WIFI_REGDOM`, this can fix some wireless issues caused by inappropriate wireless radio settings.

To use that option, create the SD card as before but add `WIFI_REGDOM="US"` (or other 2-letter country code as listed in `/etc/conf.d/wireless-regdom`). You may also configure the file manually post-install by uncommenting the appropriate line for your country or region and rebooting the Pi.

One thing not in the documentation is that you may need to eject the drive using the Windows Explorer and wait 10 seconds for the SD card to finish wear leveling tasks to avoid incomplete writes or file corruption. If you haven't seen any problems since the initial issue, please let us know what you feel needs to be fixed, be it the image or the installation instructions and we will take a look at it to clarify things if necessary. Thank you.

sharicov, 2021-02-16:
Unfortunately I cannot do it right now, SD card is in the raspberry and I
am working with it. Another one just appeared to be dead. I don't think is
character's problem otherwise it wouldn't work for the second attempt.
However Balena Etcher is giving error at the end of image write, maybe
that's the reason.

sharicov, 2021-03-31:
[pikvm.txt](https://github.com/pikvm/pikvm/files/6236939/pikvm.txt)
Better late than never, attaching the file made in windows with notepad.

sharicov, 2021-03-31:
Checked the same pi zero w with retropi, wifi works.

bbeaudoin, 2021-03-31:
The keyboard won't work on boot due to the USB port being set to peripheral mode, you would need to update the config.txt to comment out `#dtoverlay=dwc2` to allow the Pi Zero to use a keyboard and mouse.

As for the wireless settings, here's my config:
```
WIFI_ESSID="ssid"
WIFI_PASSWD="password"
WIFI_REGDOM="US"
FIRSTBOOT=1
```
To my knowledge the REGDOM setting that configures /etc/conf.d/wireless_regdom is not yet active but that may be part of the issue you are having. If your region has restrictive settings, it's possible the "compatible" setting is on the wrong frequency and channel for your wireless router. Other Pi operating systems force the regulatory domain to be set before enabling wifi to avoid issues. You may need to configure that manually if it is not set properly by uncommenting your country code and rebooting.

Since you have mixed results, I expect it's partly to do with the regulatory domain setting, sometimes the Pi can get a Wifi signal without that but it may not be as strong or as stable if no country is selected.

Unfortunately I haven't been able to reproduce the issue at all, note the wireless will not be joined on the first boot, the recommended step is to boot the Pi, wait 10 minutes (or at least until it's reached full runlevel), and unplug/replug (reboot) the Pi so wireless will be configured. If you enabled keyboard support, be sure to disable keyboard support again by enabling the dwc2 overlay for mode=peripheral (just uncomment the line again) and things should be as normal with no functioning physical keyboard as the port is used to simulate a keyboard for the host.

sharicov, 2021-04-01:
Thanks a lot for the detailed answer, unfortunately WIFI_REGDOM didn't help.
I am waiting 10 minutes every time. Will try with keyboard later on

sharicov, 2021-04-01:
All fixed, thanks to bbeaudoin's help, thank you very much!
After connecting keyboard and networking adapter I had successfully created wif profiles to all necessary networks using following guide: https://github.com/pikvm/pikvm/blob/master/pages/wifi_config.md
and installed tailscale using this guide: https://github.com/pikvm/pikvm/blob/master/pages/tailscale.md
All works now (I had to reboot my router once, though in order to connect)
Thanks again!!!

closed 2021-04-01 by mdevaev