pikvm/pikvm #195: pikvm as a SPICE server, usb redirection

1ucir, 2021-02-15:
Is it possible pikvm will be able to act not only as a VNC but also as a [SPICE](https://www.spice-space.org/) server? I am especially interested in using a spice feature that allows [redirecting](https://www.spice-space.org/spice-user-manual.html#_usb_redirection) an arbitrary usb device plugged in on the client side through the network to the pikvm-controlled pc.

mdevaev, 2021-02-15:
Good idea, but we'll have to write a server or adapt an existing one.

kevinmccarthy, 2022-03-12:
The packaging of VirtualHere (static binaries, no kernel code) may be more convenient. Has anybody tried this option?

https://www.virtualhere.com/usb_server_software

mdevaev, 2022-03-12:
This is a proprietary solution, I can't use it in the mainline.

mdevaev, 2022-07-24:
[SPICE is deprecated by Red Hat](https://linux.web.cern.ch/rhel/releasenotes/Red_Hat_Enterprise_Linux-8-8.6_Release_Notes-en-US.pdf)

> **SPICE has been deprecated**
> The SPICE remote display protocol has become deprecated. Note that SPICE will remain supported in
> RHEL 8, but Red Hat recommends using alternate solutions for remote display streaming:
> * For remote console access, use the VNC protocol.
> * For advanced remote display functions, use third party tools such as RDP, HP RGS, or Mechdyne TGX.

mdevaev, 2022-07-26:
Please follow the new task with WebUSB: #672 since it will be the preferred implementation method.

closed 2022-07-26 by mdevaev