pikvm/pikvm #197: PizeroW Web Password

Testato, 2021-02-16:
I used the pizerow image from download page to flash a microsd, so i started from clean system. 
After a dual reboot I entered by chrome on https://192.168.2.200/extras/webterm/ttyd/

On your Flashing Info page https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md
you writed:
> To change Pi-KVM web password use kvmd-htpasswd set admin. As indicated on the login screen, you need to use rw to make the root filesystem writable before issuing these commands. 

But there is some permission problem
[kvmd-webterm@pikvm ~]$ rw
mount: /: must be superuser to use mount.
mount: /boot: must be superuser to use mount.

mdevaev, 2021-02-16:
```
su -
<enter root password>
rw
```

Testato, 2021-02-16:
ok, it work
so it is better add this info on https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md

mdevaev, 2021-02-16:
Done

closed 2021-02-16 by mdevaev