pikvm/pikvm #201: IP Restrictions

staze, 2021-02-20:
Would be good to have an easy way to set IP restrictions of who can connect to the pikvm, both via web GUI and SSH. 

I assume using something like iptables. 

Sadly, some environments don't have much in the way of firewalls blocking internet based traffic, so would rather not let anyone just hammer away at these things. =)

bbeaudoin, 2021-02-20:
The default rules are empty, if you would like to define a set of rules, you can save them to `/etc/iptables/iptables.rules` for the base Arch distribution and enable iptables using `systemctl enable iptables`.

bbeaudoin, 2021-02-20:
Thinking on this a little more, there are web-based UIs for maintaining firewall rules on a system but these aren't written to support a read-only filesystem. Web UIs are available through the Arch repos and can be deployed but using these requires the SD card be read-write and require root escalation to make the firewall changes.

Writing a fixed set of rules on a read-only filesystem that are established on boot, if desired. For remote access, use of a VPN or other secure remote access solution, may be recommended and only the VPN service port exposed for administration.