pikvm/pikvm #202: Ability to pin macros/shortcuts to the UI

bionemesis, 2021-02-23:
When using a compatible HDMI KVM (like https://www.aliexpress.com/item/4001230679168.html?spm=a2g0s.9042311.0.0.354b4c4dCzrzLU), it would be helpful to have the ability to pin the keyboard shortcuts that are required for switching between inputs directly in the UI. I would see this working similar to how the GPIO menu works on the ezCoo. Perhaps an "Input" button that then shows 1/2/3/4 and when clicking on them, it sends the required keystrokes through to the switch.

jamesmills1981, 2021-05-07:
+1 for this feature, having the ability to map in macros for "Win + P" to change screen mirroring settings is my particular use case, but power key support may also be a nice-to-have
from wikipedia -> 
e05e - power, e05f - sleep, e063 - wake

mdevaev, 2021-07-14:
This is asked very often, so I will take care of it. I combine the problem with another similar one, since the solution will be common: #207

closed 2021-07-14 by mdevaev