pikvm/pikvm #203: Support for 3rd party managed power (Wattbox)

bionemesis, 2021-02-23:
Many IT service providers, who I believe to be a great target demographic for this product and the upcoming commercial implementation, use Wattbox devices to remotely control the power for essential devices at client sites. I would see great value in supporting the documented Wattbox API to be able to send commands to a Wattbox that is on the same network as the pikvm. For us, we are just starting to use a pikvm connected to a 4-port HDMI KVM at our office for working on client PCs remotely. If supported, we could connect those same devices to a Wattbox and use pikvm to power cycle a hung machine. The Wattbox API is documented here: https://www.snapav.com/wcsstore/ExtendedSitesCatalogAssetStore/attachments/documents/PowerManagement/SupportDocuments/Wattbox%20API%20v2.0.pdf.

mdevaev, 2021-03-01:
It would be extremely difficult for me to do this without having a device in my hands. Even if I had it, I would not be able to use it, since the network in my country uses 220 volts. If you know how to write in python, I could help you with the implementation of the plugin and I am ready to take it to the upstream.

bionemesis, 2021-03-01:
Sorry, no good with python. I'll see if I can find someone locally. I have access to two Wattboxes, so I can easily hand one over. I'll check Upwork to see if there is a python programmer in my area that I could foot the bill for writing the code.

mdevaev, 2021-03-01:
How urgently do you need it? I have one developer who could do this for money (if you provide him with remote access). Also tell us what functionality you need.

mdevaev, 2021-03-03:
Sup?

mdevaev, 2021-05-29:
No response from the author of the issue.

closed 2021-05-29 by mdevaev