pikvm/pikvm #206: ezCoo multiport KVM switch always switches to Port 1 when ezcoo.py initialized

hofnarr132, 2021-02-24:
ezCoo multiport KVM switch always switches to Port 1 when ezcoo.py initialized.

Line 140 has # FIXME: Get actual state without modifying the current.

ezCoo serial port command to check which video source is connected is:
   GET OUT1 VS

response will be:
OUT1 VS INx<cr><lf>

replace INx with IN1 for port 1 (channel 0), IN2 for port 2 (channel 1) etc.

mdevaev, 2021-02-24:
To be honest, I don't remember why I did it this way, there was some problem related to it.

bozserz, 2021-03-08:
The command GET IN0 SIG STA will report all 4 input line statuses (4 lines in total, ending with 1 as online, 0 as offline), it seems, the plugin does not show this (LED is showing the selection only?)
Regards: Zoltan.

mdevaev, 2026-01-07:
It's finally fixed.