pikvm/pikvm #207: Add support for changing HDMI KVM switch input via hotkey

T-REX-XP, 2021-02-25:
Hi! It would be nice when we can define a config for changing HDMI KVM switch input via hotkey
As you know for some devices we switch the input via pressing CTRL+CTRL+1/2
Can you add the ability to emulate keypress buttons ??
Thanks

mdevaev, 2021-02-25:
> CTRL+CTRL+1/2

Is it "left + right ctrl" or press-release, press-release?

T-REX-XP, 2021-02-25:
Hi @mdevaev , it's **press-release, press-release**

> How to Switch Between Two HDMI Input Devices 
> (1)Press the Switch Button on the KVM box.
> (2)Use Keyboard Hotkey--Double click Ctrl Button(or Scroll Button), then click "1" on Numeric Keypad to switch to PC1 or click "2" to PC2.
> (3)Auto Scan Switch--Double click Ctrl Button(or Scroll Button), then click "S" to start the Auto Scan Switch Mode. Available to press any button to end it. Besides, you can set up auto-switch timers between 10s and 999s, which is great for surveillance.
[Origin](https://www.ugreen.com/products/2-in-1-out-kvm-hdmi-switch-box)

sobake, 2021-03-31:
@T-REX-XP Which KVM switch are you using? I'm planning to buy one but don't know which one works with hotkeys yet. Also, have you tried Mass Storage device via the KVM switch yet? Does it work with the switch?

T-REX-XP, 2021-03-31:
> @T-REX-XP Which KVM switch are you using? I'm planning to buy one but don't know which one works with hotkeys yet. Also, have you tried Mass Storage device via the KVM switch yet? Does it work with the switch?

I have bought  ugreen, hdmi version:
https://a.aliexpress.com/_mPfSAo3
But it is quite expensive. About 37$. It case you have vga on your servers , you can try to buy vga version. 

No, I didn't try to use mass storage device via kvm. Let me check it tomorrow.

theboyknowsclass, 2021-04-09:
I'm using a ConnectPro KVM which uses a configurable hotkey 
double click Scroll Lock
double click Caps Lock 
double click Ctrl
double click Shift

this would be great if you could implement!

hanulis, 2021-05-27:
My KVM switch support hotkey with double click Caps Lock+1/2/3/4.

I tried all USB ports but no luck. It only works when connect keyboard directly.

mdevaev, 2021-07-14:
More usecases: #202.

ghjklw, 2021-11-26:
Hi!

I got an ATEN CS84U KVM switch and can switch between the sources using the hotkey shortcuts (through the virtual keyboard), so I'm also very keen on such a feature. I think this could help broadening significantly the range of compatible KVM switches (in particular if one wants a VGA option).

I started looking into it and thought of 2 ways to implement it:
- Adding keyboard shortcuts to the Web UI, as discussed above, through some configuration like:
  ```yaml
  web:
    shortcuts:
      - name: Input 1
         sequence: ScrollLock ScrollLock Digit1 Enter
      - name: Input 2
         sequence: ScrollLock ScrollLock Digit2 Enter
  ```
  My understanding is that if we use these data to dynamically create the shortcut list in [web/kvm/navbar-shortcuts.pug](https://github.com/pikvm/kvmd/blob/dbd643b2ae8fc0b0b9745bbb07dce669566f354c/web/kvm/navbar-shortcuts.pug), no further code would be required as this is all handled through a JS script.
-  Creating a pseudo GPIO driver, looking a bit like the ezCoo one, but that instead of using a serial connection would trigger keyboard events. This might provide a slightly cleaner user experience. It would also be a more consistent implementation across the devices used - for example if some enhancements are done later to the UI to switch between the sources, this implementation would also benefit from it.

I'd be happy to help but have no experience with this project yet so I might need a bit of guidance to decide between the 2 options above and get started. Is there any developer documentation available?

Thanks!

djarbz, 2022-03-17:
Has there been any progress on this? I have a KVM that uses `ScrollLock ScrollLock ##` to switch inputs.

mdevaev, 2022-03-18:
No, sorry.

pljakobs, 2022-12-18:
I second this request. Would be great to have configurable shortcuts - not necessarily runtime configurable as I appreciate the security concious approach.

mdevaev, 2023-04-27:
Note: add ability to pin shortcut on the menu bar: #1003 

> As I use PiKVM predominantly with Domain Joined Windows machines, so use CTRL+ALT+DEL quite a lot and would appreciate option to have shortcut sit on the top bar at all times.
Or maybe add option to add custom shortcuts to the top bar, like I could do with CTRL+ALT+END too for RDP sessions.

dj08, 2023-07-13:
+1 for this feature request, guided here by the discord community. I would love to pin difficult to send keyboard shortcuts like Win+Shift+S without going into full screen mode for complete keyboard capture.

nel0x, 2023-07-28:
+1 from me as well. Especially since this is the only way to switch between KVM devices with PiKVM V4 Mini - as it has no USB host support.
I like @ghjklw's pseudo GPIO driver idea a lot.

Syco54645, 2023-09-25:
+1, this feature would open things for a lot of people.

paucapo, 2023-11-11:
As a temporary solution the file at /usr/share/kvmd/web/kvm/index.html can be edited and around line 767 add some custom shortcuts:
```HTML
<hr>
<div class="buttons-row">
   <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit1">&bull; Screen 1</button>
   <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit2">&bull; Screen 2</button>
</div>
<div class="buttons-row">
   <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit3">&bull; Screen 3</button>
   <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit4">&bull; Screen 4</button>
</div>
```

This sample is for a 4 ports KVM using ScrollLock ScrollLock 1-4 shortcuts. The ScrollLock needs to be duplicated because of the dual state.

It will be great to have it as a configuration option, but having to open the virtual keyboard each time was a hassle.

plia7, 2024-02-01:
> As a temporary solution the file at /usr/share/kvmd/web/kvm/index.html can be edited and around line 767 add some custom shortcuts:
>
> ```HTML
>
> <hr>
>
> <div class="buttons-row">
>
>    <button [...]

Thanks for your suggestion, will it be possible to use this method to make some virtual keyboard buttons to show like in this example:

[image]

slendermon, 2024-09-27:
Adding onto paucapo's answer, if you want the macro as a nav bar @plia7, here's the code (i know i'm late)

/usr/share/kvmd/web/kvm/index.html #PUT AT LINE 842
```
        <!-- ==== Need to make javascript delete class ".menu" for these elements ==== -->
      <li class="right" id="shortcuts-dropdown">
        <a class="menu menu-button row50" data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit1">First</a>
      </li>
      <li class="right" id="shortcuts-dropdown">
        <a class="menu menu-button row50" data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit2">Second</a>
      </li>
        <!-- ========================================================================= -->
```
/usr/share/kvmd/web/share/js/tools.js #PUT AT END OF FILE
```
/* =========== Custom Code ============= */

function remenu() {
    // const customacro =  document.querySelector('.menu .menu-button .row50');
    const customacro =  document.querySelector('li#shortcuts-dropdown a.menu.menu-button.row50');
    if (customacro) {
        customacro.classList.remove('menu');
        console.log('Class "menu" removed from element:', customacro);
    } else {
        console.error('Element with "li#shortcuts-dropdown a.menu.menu-button.row50" not found');
    }
}

// Wait for the DOM to fully load before setting the timeout
document.addEventListener('DOMContentLoaded', function() {
    // Call function immediately
    remenu();
});

/* ===================================== */
```

plia7, 2024-09-27:
> Adding onto paucapo's answer, if you want the macro as a nav bar @plia7, here's the code (i know i'm late)
> 
> /usr/share/kvmd/web/kvm/index.html #PUT AT LINE 842 (This will show up AFTER shortcuts, not GPIO because i dont have gpio)
> 
> ```
>         
>       <li class="right" id="shortcuts-dropdown">
>         <a class="menu menu-button row50" data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit1">Second</a>
>       </li>
>       <li class="right" id="shortcuts-dropdown">
>         <a class="menu menu-button row50" data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit2">First</a>
>       </li>
>         
> ```
> 
> /usr/share/kvmd/web/share/js/tools.js #PUT AT END OF FILE
> 
> ```
> /* =========== Custom Code ============= */
> 
> function remenu() {
>     const customacro =  document.querySelector('li#shortcuts-dropdown a.menu.menu-button.row50');
>     if (customacro) {
>         customacro.classList.remove('menu');
>         console.log('Class "menu" removed from element:', customacro);
>     } else {
>         console.error('Element with "li#shortcuts-dropdown a.menu.menu-button.row50" not found');
>     }
> }
> 
> // Wait for the DOM to fully load before running function
> document.addEventListener('DOMContentLoaded', function() {
>     // Call function immediately
>     remenu();
> });
> 
> /* ===================================== */
> ```
> 
> Result: ![image](https://private-user-images.githubusercontent.com/97182340/371643713-a63f5aae-e894-40da-8213-3716a3e39d7c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0NjMyMTQsIm5iZiI6MTcyNzQ2MjkxNCwicGF0aCI6Ii85NzE4MjM0MC8zNzE2NDM3MTMtYTYzZjVhYWUtZTg5NC00MGRhLTgyMTMtMzcxNmEzZTM5ZDdjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI3VDE4NDgzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFjYzBlYjFjMmY0OTk1M2MyMzQ3MTBmNDhmMWU5YjNiMTA1ODY4MWM3ZjUxYTE0MTM5MWIzODY1YjIwNjk2ODcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.544lQiTq8plgA4BxfT3_coabGFgs-XHgIaoHOFKqwz8) As you can see there are two new buttons: "First" and "Second"

Thanks I will give it a try!

slendermon, 2024-09-27:
Actually I was wrong, it just needs html changes

/usr/share/kvmd/web/kvm/index.html #PUT AT LINE 842 (again)
```
        <!-- ==== Need to make javascript delete class ".menu" for these elements ==== -->

      <li class="right" id="shortcuts" style="height: 50px; line-height: 50px; outline: none; cursor: pointer; display: inline-block; color: var(--cs-navbar-default-fg); padding-left: 16px; padding-right: 16px; text-decoration: none; position: relative;">
        <a class="row50" data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit1">Second</a>
      </li>
      <li class="right" id="shortcuts" style="height: 50px; line-height: 50px; outline: none; cursor: pointer; display: inline-block; color: var(--cs-navbar-default-fg); padding-left: 16px; padding-right: 16px; text-decoration: none; position: relative;">
        <a class="row50" data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit2">First</a>
      </li>

        <!-- ========================================================================= -->
```

kinda lazy, but possible to just make a css class, and put all styles there.

mark-bakermd, 2024-12-14:
> As a temporary solution the file at /usr/share/kvmd/web/kvm/index.html can be edited and around line 767 add some custom shortcuts:
> 
> ```
> <hr>
> <div class="buttons-row">
>    <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit1">&bull; Screen 1</button>
>    <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit2">&bull; Screen 2</button>
> </div>
> <div class="buttons-row">
>    <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit3">&bull; Screen 3</button>
>    <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit4">&bull; Screen 4</button>
> </div>
> ```
> 
> This sample is for a 4 ports KVM using ScrollLock ScrollLock 1-4 shortcuts. The ScrollLock needs to be duplicated because of the dual state.
> 
> It will be great to have it as a configuration option, but having to open the virtual keyboard each time was a hassle.

This is fantastic. Thank you!
I think the next step for me is going to be to try and wire into gpio to use the serial interface on the KVM. It's unfortunate that the ports which support the ScrollLock shortcut do not support USB storage devices, limiting one of the coolest features of PiKVM.

mark-bakermd, 2024-12-14:
This is where I landed for the 16-port model (only using the first 8 right now)
```
            <hr>            
            <div class="buttons-row">
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit1">&bull; Screen 1</button>
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit2">&bull; Screen 2</button>
            </div>       
            <div class="buttons-row">
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit3">&bull; Screen 3</button>
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit4">&bull; Screen 4</button>
            </div>  
            <div class="buttons-row">
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit5">&bull; Screen 5</button>
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit6">&bull; Screen 6</button>
            </div>       
            <div class="buttons-row">
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit7">&bull; Screen 7</button>
               <button class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit8">&bull; Screen 8</button>
            </div> 
```

dragon788, 2025-01-21:
> Sorry, actually I was wrong, it just needs html (and css) changes [@plia7](https://github.com/plia7)
> 
> /usr/share/kvmd/web/kvm/index.html #PUT AT LINE 842 (again)
> 
> ```
>         
> 
>       <li class="navi right" id="shortcuts">
>         <a class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit2">Second</a>
>       </li>
>       <li class="navi right" id="shortcuts">
>         <a class="row50" data-force-hide-menu data-shortcut="ScrollLock ScrollLock ScrollLock ScrollLock Digit1">First</a>
>       </li>
> 
>         
> ```
> 
> /usr/share/kvmd/web/share/css/navbar.css #END OF FILE
> 
> ```
> .navi {
>     height: 50px;
>     line-height: 50px;
>     outline: none;
>     cursor: pointer;
>     display: inline-block;
>     color: var(--cs-navbar-default-fg);
>     padding-left: 16px;
>     padding-right: 16px;
>     text-decoration: none;
>     position: relative;
> }
> 
> .navi:hover:not(.active),
> a.navi:hover:not(.active) {
>     background-color: var(--cs-navbar-item-hovered-bg);
> }
> ```
> 
> the last workaround produced errors in console, and couldnt move keyboard, this fixes it, fixes all errors from console!

I wonder if it would be possible to create a git diff "patch" file that applies this regardless of the specific lines by looking at the surrounding context so it could be applied cleanly as long as this section doesn't change significantly between versions.

mdevaev, 2025-10-21:
I'm working on improving the UI features, so it's likely that this feature will appear at some point.

mdevaev, 2025-10-26:
Note: this feature should provide an ability to make shortcuts like ScrLk[press+release]

mdevaev, 2025-11-09:
This is not exactly what is being asked for here, but I fixed another issue related to the inconvenience of entering hotkeys. Please take a look, some of your usecases should be solved there: https://docs.pikvm.org/shortcuts