pikvm/pikvm #208: Permanent image

sharicov, 2021-02-25:
Is it possible to make selected ISO image permanent even after Raspberry power off?
Thanks

mdevaev, 2021-02-25:
No. Might be implemented.

mdevaev, 2021-03-01:
BTW why you need this?

sharicov, 2021-03-01:
Laptop is booting faster than raspberry zero. in order to boot from iso placed in raspberry need to turn on the laptop, enter bios (so USB port is powered and laptop doesn't boot), wait until pikvm boots, than connect to raspberry from another computer, than rebooting the controlled laptop.
Would be much faster if ISO would load automatically on pikvm boot...
Thank you for doing great job! Appreciate your time and support!

bozserz, 2021-05-10:
Hello, just a quick idea: there are cheap USB power banks out there. On the Pi Zero it's especially easy to connect both usb-s simultaneously, as there are 2 connectors. The connector close to the HDMI should be connected to the computer, and the one closer to the edge can go to a power bank or other power source (for the mobility I'm using the first). You can then reboot the laptop as many times You want, it will stay there with the ISO already mounted... On the Pi 4 a special cable or power injector board unit is needed.  However having a fix CD ISO mounted is a good feature request.
Similar idea: (it does not apply to ZeroW) as the Pi4 can now boot from a USB drive, it could be also possible to have a partition on the USB drive that can be easily accessed via any other PC (even Windows) and have a fix Flash disk even in R/W mode mounted... either booted the PiKVM from an SD or an other partition of the USB drive... In case of corruption/error there, the data can be easily re-written, the system will still be operational, as it's a separate partition (or even device).

theLMGN, 2021-05-23:
Machines that I've tested leave the USB ports powered if you reboot instead of shutting down and starting up again. Not ideal, but hopefully helpful?

mdevaev, 2021-05-24:
Will be available since 2.78. Configuring like this:
```yaml
kvmd:
    msd:
        initial:
            image: os-image.iso
            cdrom: true
```

closed 2021-05-24 by mdevaev

mdevaev, 2021-05-25:
Released

sharicov, 2021-05-26:
Thanks for implementing the feature.
Getting following error:
journalctl -u kvmd

```
May 19 19:35:56 pikvm systemd[1]: Started Pi-KVM - The main daemon.
May 26 10:13:56 pikvm kvmd[370]: ConfigError: Invalid value '/var/lib/kvmd/msd/images/Sharicov_mini_PE.iso' for key 'kvmd/msd/initial/image': The argument '/var/l>
May 26 10:13:57 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
May 26 10:14:03 pikvm kvmd-cleanup[452]: kvmd.apps.cleanup                 INFO --- Cleaning up ...
May 26 10:14:03 pikvm kvmd-cleanup[452]: kvmd.apps.cleanup                 INFO --- Trying to find and kill the streamer 'kvmd/streamer: <app>' ...
May 26 10:14:04 pikvm kvmd-cleanup[452]: kvmd.apps.cleanup                 INFO --- Bye-bye
May 26 10:14:04 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
May 26 10:14:04 pikvm systemd[1]: kvmd.service: Consumed 19.189s CPU time.
May 26 10:14:07 pikvm systemd[1]: kvmd.service: Scheduled restart job, restart counter is at 1.
May 26 10:14:07 pikvm systemd[1]: Stopped Pi-KVM - The main daemon.
May 26 10:14:07 pikvm systemd[1]: kvmd.service: Consumed 19.189s CPU time.
May 26 10:14:07 pikvm systemd[1]: Started Pi-KVM - The main daemon.
May 26 10:14:20 pikvm kvmd[479]: ConfigError: Invalid value '/var/lib/kvmd/msd/images/Sharicov_mini_PE.iso' for key 'kvmd/msd/initial/image': The argument '/var/l>
May 26 10:14:20 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
May 26 10:14:26 pikvm kvmd-cleanup[481]: kvmd.apps.cleanup                 INFO --- Cleaning up ...
May 26 10:14:26 pikvm kvmd-cleanup[481]: kvmd.apps.cleanup                 INFO --- Trying to find and kill the streamer 'kvmd/streamer: <app>' ...
May 26 10:14:26 pikvm kvmd-cleanup[481]: kvmd.apps.cleanup                 INFO --- Bye-bye
May 26 10:14:27 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
May 26 10:14:27 pikvm systemd[1]: kvmd.service: Consumed 18.705s CPU time.
May 26 10:14:30 pikvm systemd[1]: kvmd.service: Scheduled restart job, restart counter is at 2.
May 26 10:14:30 pikvm systemd[1]: Stopped Pi-KVM - The main daemon.
May 26 10:14:30 pikvm systemd[1]: kvmd.service: Consumed 18.705s CPU time.
May 26 10:14:30 pikvm systemd[1]: Started Pi-KVM - The main daemon.
```

Please assist