pikvm/pikvm #212: No stream

sharicov, 2021-02-25:
Just connected lusya (black) adapter to raspberry zero w, getting no signal.
Red led on the lusya, PIKVM works, but no video.
How can I make sure it's HW fault?
dmsg:
[   19.035294] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
[   19.149494] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   19.198704] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
[   19.268002] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   19.360427] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
[   19.388320] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   19.507152] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   19.573128] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
[   19.654552] i2c i2c-11: Added multiplexed i2c bus 0
[   19.691900] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   19.766646] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[   19.816514] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   19.999438] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller

mdevaev, 2021-02-25:
Attach `journalctl -u kvmd`. Please use markup or attach it as file.

Please, with similar problems, contact the discord chat. Most of what you are experiencing is the result of incorrect configuration or the problem has already been solved by other users. Github is intended for error messages or feature requests, and in rare cases, for questions. There you will be able to answer not only me, but also other users, and they will do it faster.

sharicov, 2021-02-25:
[hdmi.txt](https://github.com/pikvm/pikvm/files/6045443/hdmi.txt)
Thanks in advance

mdevaev, 2021-02-25:
> Persistent device timeout (unplugged)

Your source is not sending a signal, or a bad HDMI cable, or a bad CSI connection. This is not a bug. With this question, you really should contact the chat.

closed 2021-02-25 by mdevaev