pikvm/pikvm #214: Change HID logical minimum to 1

wisteso, 2021-02-28:
I noticed that the min value was changed to 0 in the HID descriptor due to what looks like windows behavior?
https://github.com/pikvm/kvmd/blob/f5246c410e8daed03052b84b222cf498843c4cd8/kvmd/apps/otg/hid/mouse.py#L82

However, clamping still happens to -32768 in this validator function:
https://github.com/pikvm/kvmd/blob/f5246c410e8daed03052b84b222cf498843c4cd8/kvmd/validators/hid.py#L47

And perhaps you should consider clamping to 1, rather than 0, since it seems that Windows discards any mouse reports which are entirely null (zero) bytes.

Testable by using something like `echo -ne \\x00\\x00\\x00\\x00\\x00\\x00 >> /dev/hidg1` and observing that the mouse doesnt change position, while `echo -ne \\x00\\x01\\x00\\x01\\x00\\x00 >> /dev/hidg1` does have the desired effect.

mdevaev, 2021-04-02:
I tried to reproduce it with same conditions (win 10 x64, rpi4 otg) but the problem doesn't seem to be reproducing. Maybe you have an old version of windows, missing some updates?

mdevaev, 2021-04-17:
@wisteso boop

mdevaev, 2021-04-24:
@wisteso hello?

mdevaev, 2021-07-24:
I can't reproduce it at all and no answer from the author.

If you want to continue researching this problem, please write here and I will reopen it.

closed 2021-07-24 by mdevaev