pikvm/pikvm #217: Elgato CamLink Unsupported ?

karepiu, 2021-03-01:
I got below with Elgato Cam Link. Any idea ? 

`[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: ================================================================================
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [87.886    stream] -- Device fd=8 opened
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [87.886    stream] -- Using input channel: 0
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [87.966    stream] -- Requested resolution=1920x1080 is unavailable
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.046    stream] -- Using resolution: 3840x2160
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [88.046    stream] -- Could not obtain the requested pixelformat=JPEG; driver gave us unsupported
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [88.046    stream] -- Unsupported pixelformat=NV12 (fourcc)
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.046    stream] -- Device fd=8 closed
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.046    stream] -- Sleeping 1 seconds before new stream init ...
[2021-03-01 09:21:20 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: ================================================================================`

mdevaev, 2021-03-01:
NV12 format is not supported. Use UYVY, YUYV, RGB24, JPEG or MJPEG.

karepiu, 2021-03-01:
Is there a way to change it from UI ?

mdevaev, 2021-03-01:
No. First you need to make sure that the camera supports the desired formats. Show `v4l2-ctl --list-formats-ext -d /dev/kvmd-video`.

karepiu, 2021-03-01:
`[kvmd-webterm@pikvm ~]$ v4l2-ctl --list-formats-ext -d /dev/kvmd-video
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'NV12' (Y/CbCr 4:2:0)
                Size: Discrete 3840x2160
                        Interval: Discrete 0.033s (29.970 fps)
        [1]: 'NV12' (Y/CbCr 4:2:0)
                Size: Discrete 3840x2160
                        Interval: Discrete 0.033s (29.970 fps)
        [2]: 'YU12' (Planar YUV 4:2:0)
                Size: Discrete 3840x2160
                        Interval: Discrete 0.033s (29.970 fps)
[kvmd-webterm@pikvm ~]$ `

mdevaev, 2021-03-01:
Sorry, but no chance to enable it. It has no supported formats.

karepiu, 2021-03-01:
That is interesting. Seems the question was correct - Elgato is not supported

Elgato seems to be power house currently for all streaming capture devices ( half of the world uses it for OBS ) - strange that so popular and unfortunately expensive hardware does not work.  Bummer

Nevertheless - I would suggest a feature - dropdown allowing to change the format in UI.

mdevaev, 2021-03-01:
In fact, I think I could add support for this thing, but it requires separate research. The reason why this is not supported is very simple: I don't have the money to buy and test such devices. Especially since this is not a good fit for Pi-KVM. This thing does not know how to compress an image on its own and compressing video on a raspberry 4k will require huge hardware resources.

mdevaev, 2021-03-01:
The format is configured once and then used repeatedly. If ustreamer supported one of the formats of this device, it would work without explicitly specifying the format on your part.

karepiu, 2021-03-01:
1. Makes sense that it does not do any compression - it is usually connected to powerful streaming PC. The question would be - is it worth to support it even though it is not good for this solution just because it is popular - hard to say. 
2. What if device supports more then one supported format ? Wouldn't allowing ability to choose from supported format be beneficial ? ( imho having control is always better than not having one )

mdevaev, 2021-03-01:
1. This device will not be able to run on a RPi as efficiently as on a PC simply because of the lack of resources on the first one.
2. The video format is a low-level setting that can be done via the config. If you look at /etc/kvmd/main.yaml, then you will see that the mjpeg format is configured there by default. This setting is sufficient for absolutely all cases of using the recommended usb hdmi grabber. Similar for hdmi csi.

Let's imagine that ustreamer supports all three formats of your. What format will you choose? NV12? And how will it differ from the other two? This is a technical setting that does not make sense to change the user at all. It doesn't affect absolutely anything, it's just the format of transmitting pixel data from the device to the streamer.

For example, CSI HDMI supports UYVY and RGB24. By default, the first one is used. The second one is never needed at all, it gives nothing and is even worse supported by the driver. Once you need to choose the best format for the device and never change it. You needed this setup because your device didn't work. If all three formats were supported, the driver would choose one of them and just use it.

karepiu, 2021-03-01:
I understand it and it is ultimately your decision and your product - so whatever you decide stands :). 

My experience though suggest that there is lots of edge cases everywhere and hiding option from user just because something happens automatically ( or as I sometimes say "automagically" ) is almost never good decision. What if I have device that supports UYVY and RGB24 but for some reason implementation of UYVY is broken. Most of power users probably would override main.yaml etc. but would have to figure it out first that this is an issue. Having dropdown would allow anybody to try different setups and choose the best for their situation - it may not work in my case as my device simply does not support needed formats but it may be great option for somebody else. Ofc it will not matter for your v3 hardware but for all DIY may be absolutely great option. 

Whatever you decide - good luck and Thank you for the great work. 

I will just purchase different capture card and try it out again.

mdevaev, 2021-03-02:
If UYVY is broken, then you will know it from the log. In Pi-KVM, there are fifty parameters that somehow affect the video. If you follow this logic, you need to ensure that they are configured from the GUI, without giving preference only to the format. You are experiencing a problem because you are using a device that is not on the recommended list. In addition, as far as I can tell, changing the format alone would not be enough. It seems that other settings need to be changed, such as the list of screen resolutions.