pikvm/pikvm #218: Question about relays

Barborica-Alexandru, 2021-03-01:
Hi, I hope i'm not a bother, I just have a quick question: the octocouplers suggested in the build guide I can't seem to find at my local retailer so I was curios if I could use this solid state relay I have on hand? I know I'm not supposed to use mechanical relays, but I never used a solid state one, maybe it's more reliable? I believe they are high activated vs low activated.
[image]

mdevaev, 2021-03-01:
Hi. You can use mechanical relays, but they must be activated by a high signal.

Barborica-Alexandru, 2021-03-01:
> 
> 
> Hi. You can use mechanical or solid relays, but they must be activated by a high signal.

Oh wow thanks for the instant reply. Also thank you for your hard work!

mdevaev, 2021-03-01:
ur welcome

closed 2021-03-01 by mdevaev