pikvm/pikvm #219: Mouse Position scaled when windows has a second display configured

f0rdprefect, 2021-03-02:
**Describe the bug**
Have a second display configured but not necessarily connected on a windows box

**To Reproduce**
Steps to reproduce the behavior, like:

Move the mouse over the screen. In my case the 2nd display was to the right of the first display. The more the mouse is moved to the right, the more the windows mouse cursor differs from the position indicated by the blue dot.

Once I disable the second screen blue dot and mouse cursor coincide.

**Expected behavior**

Not sure. Could be hard to detect the host machine display layout. Maybe this needs to be a known issue?

**Desktop (please complete the following information):**
 - OS: Win10
 - Browser chrome
 - Version 88
 - VNC client (if used)

**Pi-KVM info:**
  - Raspberry Pi board version RPi 4
  - Pi-KVM platform v2-hdmi
  - Video capture type CSI bridge
  - KVMD version: 2.24
  - uStreamer version: 3.16
  - Linux kernel: 5.10.14

**Additional context**
Add any other context about the problem here.

mdevaev, 2021-03-02:
This is a known multi-monitor configuration issue and there is no way to fix it right now. However, there is a workaround. You can use the relative mouse mode: https://github.com/pikvm/pikvm/blob/master/pages/mouse.md

mdevaev, 2021-03-06:
The question is answered, don't hesitate if you want to discuss something else.

closed 2021-03-06 by mdevaev

andrelung, 2024-06-14:
correct link:
https://github.com/pikvm/pikvm/blob/master/docs/mouse.md