pikvm/pikvm #220: Question: Cabling RPI again

polachz, 2021-03-03:
Hallo. 

I have found the project yesterday and I'm impressed by the idea. But some things are not completely clear for me. Especially with connecting RPI to  the server as keyboard/mouse

I have found the issue #11 but it didn't give me all answers. Maybe I do not understand RPI USB system completely. Can you please answer to me several questions? 

1. Why on older RPI (3 and bellow) I can't use an USB/A  port as keyboard or mouse emulator directly? Why I need to use Arduino here instead just connect RPI USB port with server USB port? 
2.  Why I need to use Spliiter cable on RPI4 instead just use usb-C port as power supply and another USB/A port as Keyboard/mouse? 
3. If the reason for 2 is that power will be provided on the USB cable(and then some servers wont boot), isn't easier solution to cut off 5V wire from the cable and then use the RPI USB/A  port to connect the server instead of provide Y splitter? Will this solution works or I'm missing something here? 

Thank you so much for your help

mdevaev, 2021-03-03:
Hello. There is a hardware difference between a USB host and a USB device. The USB-A ports on the Raspberry can only work in host mode. They are not able to emulate the device. All the other problems stem from here.

polachz, 2021-03-03:
Thank you for the explanation! By quick googling with the provided knowledge  I have found that to get what I want is possible only on RPI model A (USB port tirectly connected to controller, not to  hub). But the RPI 3A (https://www.raspberrypi.org/products/raspberry-pi-3-model-a-plus/) has no enough power to provide KVM, right?

mdevaev, 2021-03-03:
This one has enough power, but it's not too convenient to use (no ethernet) and I don't support this device out of the box.

polachz, 2021-03-03:
Ok. Thank you so much!  It's enough from me. You can close the issue now.

mdevaev, 2021-03-03:
Welcome

closed 2021-03-03 by mdevaev