pikvm/pikvm #221: unable to enumerate USB device

torvitas, 2021-03-03:
**Describe the bug**
USB HID Device "does not work" on server.

**To Reproduce**
Steps to reproduce the behavior, like:
1. connect hdmi and power to pikvm
1. on server: dmesg -w
2. insert usb to server
4. See error

**Expected behavior**
HID device works.

**Desktop (please complete the following information):**
 - OS: Fedora 33
 - Browser Firefox 85

**Pi-KVM info:**
  - RPi4
  - v2-hdmi
  -  CSI bridge
  - kvmd 2.29-1
    kvmd-oled 0.3-1
    kvmd-platform-v2-hdmi-rpi4 2.29-1
    kvmd-webterm 0.36-1
  - uStreamer version: ustreamer 3.17-1
  - Linux kernel: Linux pikvm 5.10.17-2-ARCH #1 SMP Tue Feb 23 22:42:50 UTC 2021 armv7l GNU/Linux

**Additional context**
```
[23722.896725] usb 1-2.2.2.1: new high-speed USB device number 54 using xhci_hcd
[23722.972847] usb 1-2.2.2.1: device descriptor read/64, error -71
[23723.152884] usb 1-2.2.2.1: device descriptor read/64, error -71
[23723.336903] usb 1-2.2.2.1: new high-speed USB device number 55 using xhci_hcd
[23723.413711] usb 1-2.2.2.1: device descriptor read/64, error -71
[23723.592891] usb 1-2.2.2.1: device descriptor read/64, error -71
[23723.700756] usb 1-2.2.2-port1: attempt power cycle
[23724.296696] usb 1-2.2.2.1: new high-speed USB device number 56 using xhci_hcd
[23724.296731] usb 1-2.2.2.1: Device not responding to setup address.
[23724.500730] usb 1-2.2.2.1: Device not responding to setup address.
[23724.708696] usb 1-2.2.2.1: device not accepting address 56, error -71
[23724.784697] usb 1-2.2.2.1: new high-speed USB device number 57 using xhci_hcd
[23724.784731] usb 1-2.2.2.1: Device not responding to setup address.
[23724.988730] usb 1-2.2.2.1: Device not responding to setup address.
[23725.196695] usb 1-2.2.2.1: device not accepting address 57, error -71
[23725.196743] usb 1-2.2.2-port1: unable to enumerate USB device
```

torvitas, 2021-03-03:
Probably cable related.

closed 2021-03-03 by torvitas

c4talyst, 2022-11-25:
Probably? Was it cable related, or was it not? Having the same issue with Ubuntu 22.10 on a Raspberry Pi 4.

mdevaev, 2022-11-25:
Please contact our discord to solve this.

torvitas, 2022-11-25:
> Probably? Was it cable related, or was it not? Having the same issue with Ubuntu 22.10 on a Raspberry Pi 4.

It was cable related. The root cause was that I used wire instead of cord.

c4talyst, 2022-11-28:
Thanks! I tried a new cable and it worked.