pikvm/pikvm #222: Table in `multiport.md` file not accessible to colorblind users

garrettheath4, 2021-03-03:
**Describe the bug**
The bottom of the [_multiport.md_](https://github.com/pikvm/pikvm/blob/master/pages/multiport.md) Markdown document that is linked from the Pi-KVM repo's main README document (the "Multiport KVM over IP" link under the _Tips_ heading) contains a table with 4 rows where the "Status" column is a small box whose color indicates whether that model of KVM works with Pi-KVM. The problem is that the colors are indistinguishable to viewers (like myself) with color vision deficiency (aka "colorblindness").

**To Reproduce**
Look at the [_multiport.md_](https://github.com/pikvm/pikvm/blob/master/pages/multiport.md) page or screenshot using a color vision deficiency simulator. There are many emulation options out there including [one built-in to Chrome v83+](https://developers.google.com/web/updates/2020/03/devtools), mobile apps, and web apps.

**Expected behavior**
You should be able to interpret all of the information in the document without having to distinguish between different colors. In other words, information can be conveyed by color but _only_ if that information is conveyed by another means as well (for example, the colorized item is also a different **shape/icon** depending on its meaning).

For more guidance please see [WCAG Guideline item 1.4.1](https://www.w3.org/WAI/WCAG21/quickref/#use-of-color) for a technical guideline or the [A11Y Project's WCAG Compliance Checklist](https://www.a11yproject.com/checklist/#make-sure-color-isnt-the-only-way-information-is-conveyed) for an easier-to-read checklist.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Catalina v10.15.7
 - Browser: Chrome 
 - Version: 88.0.4324.192
 - VNC client: _n/a_

**Pi-KVM info:**
_n/a_

mdevaev, 2021-03-03:
Thanks for the information. I'll take care of it.

mdevaev, 2021-03-06:
Fixed using special characters.

closed 2021-03-06 by mdevaev