pikvm/pikvm #225: Document backpower prevention on Pi Zero W

daxelrod, 2021-03-06:
**Describe the bug**
Per @bbeaudoin https://discord.com/channels/580094191938437144/580094193834131456/816658993375084554 , the Pi Zero W can receive power from the PC over the OTG port just like the Pi 4.

While the [hardware list for v2](https://github.com/pikvm/pikvm/#hardware-for-v2) describes how to make a Pi 4 splitter that prevents backpower from the PC, for the Zero W, it just recommends "2x USB A-to-micro cables (male-male, for power and keyboard & mouse emulator)."

I have been unable to find documentation from the Raspberry Pi Foundation about whether it is harmful for the Pi Zero W to receive power from both the PC and a power supply simultaneously. If it is harmful, please modify the instructions to suggest a data-only USB-A to micro cable, or describe how to modify a cable.

Thank you!

**To Reproduce**
Read https://github.com/pikvm/pikvm/#hardware-for-v2

**Expected behavior**
If it is harmful for a Pi Zero W to receive power on both the OTG port from the PC and from a power supply on the power port simultaneously, the documentation should provide instructions on mitigating this.

**Pi-KVM info:**
  - Raspberry Pi board version: `zerow`
  - Pi-KVM platform: `v2-hdmi`
  - Video capture type: CSI bridge

Testato, 2021-03-06:
I can confirm the problem.
I cutted the V+ cable on a usbA to microUsb cable

wnark, 2021-03-08:
In addition to the conventional microusb interface, there is also GPIO power supply and OTG power supply at the same time, will it burn the Raspberry Pi?

Testato, 2021-03-08:
Every kind of "dual" supply should be  avoided.

wnark, 2021-03-08:
> Every kind of "dual" supply should be avoided.

I thought of this because the Raspberry Pi battery is powered by GPIO, haha, what you said is indeed correct.

bbeaudoin, 2021-03-08:
Whether differences between supply voltage and host voltage will damage the Pi Zero W (or Pi 4, for that matter), the Pi does not have power protection circuitry to prevent a stronger voltage source from ovewhelming the supply or the host. This issue is not one unique to the Pi Zero W, it is for all Raspberry Pi hardware and is documented.

1. https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf -- External Schottky circuit to prevent backpower from affecting host or supply or to OR the two supplies so the Pico will be powered by the stronger of two sources.
2. https://datasheets.raspberrypi.org/cm4io/cm4io-datasheet.pdf -- jumper on the board to prevent backpower when the device will be powered by a separate supply.
3. https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md -- General warning about devices that do not provide a diode to protect against backpower possibly damaging the Pi.
3. https://hackaday.com/2019/07/16/exploring-the-raspberry-pi-4-usb-c-issue-in-depth/ -- More information about USB-C power and the backpower issue causing reliability issues when receiving both supply and host voltage
4. https://www.raspberrypi.org/forums/viewtopic.php?t=249877 -- Using a power injector with a device in OTG mode (specific to the Pi 4 but can also be applied generally to any Pi receiving power from multiple sources.

This is not a special use case, it's typical for all Raspberry Pi models that support OTG mode. When the Zero is connected to power, the supply also supplies the USB peripherals, most commonly keyboard and mouse. When the Zero is connected to a host via the USB port, it is expected to be in gadget (OTG) mode and only receive power from the host. In such a case where the Pi Zero W must be connected to both a supply and a host, the recommendation is not to connect the +5V to the host.

https://www.raspberrypi.org/forums/viewtopic.php?t=223891

Ultimately yes, backpower can not only damage the Pi Zero W but any device or connected supply to which backpower protection circuitry is not implemented. There are cables called "PC Transfer Cables" that can connect two hosts, USB cables that are data-only, power/data splitters, some Y cables, and other products available that solve for this issue.

While only the Raspberry Pi Compute Module 4 I/O board provides a specific feature to prevent backpowering hosts, it does not provide a complete circuit (only a jumper that will connect or disconnect power), the Raspberry Pi Pico datasheet contains schematics for those who wish to build partial or full backpower protection. Otherwise it's simple enough just to recommend not to connect power between the Pi and the host unless the Pi will be host powered and an external supply will not be used.

mdevaev, 2021-03-09:
@bbeaudoin I propose to supplement the point in zero with the need to cut off the power wire in the cable or seal the corresponding contact in the connector.

bbeaudoin, 2021-03-09:
> @bbeaudoin I propose to supplement the point in zero with the need to cut off the power wire in the cable or seal the corresponding contact in the connector.

In the event the Pi is plugged into two 5V sources I agree. The only safe way to power the Pi from a secondary external source is to disable the 5V connection on the OTG cable connecting the Pi to the host.

mdevaev, 2021-03-09:
Done.

closed 2021-03-09 by mdevaev