pikvm/pikvm #226: Use of lets encrypt for fully qualified domain name

animeai, 2021-03-07:
**Is your feature request related to a problem? Please describe.**
I've not had a chance to try this software out yet as I'm waiting on a delivery from China. From reading the documentation this uses a self signed certificate. I'm planning on locking it down to local IPs only and using my current docker with traefik setup as a reverse proxy to enable a sub-domain to work from outside of my network with a full lets encrypt certificate. This however would cause issues since it is the docker server I want the KVM on in case I can't access SSH. Setting up IP tables to restrict access isn't an issue but I hate self signed certificates when I have everything else on my network running on a very short domain name!

**Describe the solution you'd like**
Addition of an optional domain/subdomain name with an SSL certificate provided by LetsEncrypt. This would however require:

1. Owned domain name with customisable DNS or use of a dynamic DNS service such as DuckDNS.
2. DNS for the domain pointed at the correct IP with correct local NAT routing to the Pi.
3. External access on port 80/443 for letsencrypt to authenticate the certificate.

**Describe alternatives you've considered**

- Manual set up of lets-encrypt and ZeroTier powered global area networking on the pi with IP tables locking access to the ZeroTier IP only.
- (Docker + traefik + letsencrypt + authelia) NGINX reverse proxy on another machine with IP tables set on the pi to lock the KVM to local IPs only. The reverse proxy would accept the self signed certificate and traefik would use its own lets encrypt certificate generation to upgrade to fully signed SSL. This is my preferred solution for personal use as everything except a custom NGINX container pointed to the correct local IP is already set up on my network.
- Use of local self-hosted Certificate Authority to sign self-signed certificates for use without a fully qualified domain name (I.E. https://pikvm.local).

sIKE23, 2021-04-15:
Requirements:
-	You own your own public domain name through a public registrar
-	You have the ability to create DNS Records for your public domain
-	The system has internet access
Note: commands that you will issue as a part of this process are bolded
First step is to install certbot; this is the utility that talks to the Lets Encrypt servers to generate the certificate for your domain. 

Login to your PI-KVM console and perform the following steps:
sudo su - 
rw
pacman -S certbot 
	Proceed with installation? [Y/n] y

Now that certbot is installed, you will need to create the command to create the certificate for your domain. In my example below, I am using the contoso.com domain. Please substitute your e-mail address and domain name to build your command to generate your certificate. 

Please note I am using the –preferred-challenges=dns switch instead of using the default method which uses an http query to a file on a webserver to verify that you own the domain in question. I have found it much easier, straight forward to use the DNS based approach, and that is what you will find below.

certbot certonly --manual --preferred-challenges=dns  --email myemail@contoso.com --server https://acme-v02.api.letsencrypt.org/directory  --agree-tos -d pikvm.contoso.com
You will be prompted with a series of questions and it should go something like this:
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator manual, Installer None

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Would you be willing, once your first certificate is successfully issued, to
share your email address with the Electronic Frontier Foundation, a founding
partner of the Let's Encrypt project and the non-profit organization that
develops Certbot? We'd like to send you email about our work encrypting the web,
EFF news, campaigns, and ways to support digital freedom.
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
(Y)es/(N)o: n
Account registered.
Requesting a certificate for pikvm.contoso.com
Performing the following challenges:
dns-01 challenge for pikvm.contoso.com

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Please deploy a DNS TXT record under the name
_acme-challenge. pikvm.contoso.com with the following value:

QRY34h618geOX15Se106zHy0pyMnHD-KgVB1SbZ3Eq2

Before continuing, verify the record is deployed.
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Press Enter to Continue

You will see that I answered no to the EFF question. I was then was prompted to create a DNS record in my DNS zone. At this point, you will need to go to web page/DNS server that manages your domain and create the record exactly as per the request.

Once you have created the DNS record you will then press Enter to Continue. If you have everything right you should see something like:
Waiting for verification...
Cleaning up challenges

IMPORTANT NOTES:
 - Congratulations! Your certificate and chain have been saved at:
   /etc/letsencrypt/live/pikvm.contoso.com/fullchain.pem
   Your key file has been saved at:
   /etc/letsencrypt/live/pikvm.contoso.com/privkey.pem
   Your certificate will expire on 2021-06-30. To obtain a new or
   tweaked version of this certificate in the future, simply run
   certbot again. To non-interactively renew *all* of your
   certificates, run "certbot renew"
 - If you like Certbot, please consider supporting our work by:

   Donating to ISRG / Let's Encrypt:   https://letsencrypt.org/donate
   Donating to EFF:                    https://eff.org/donate-le

Yaay! You now have certificates for domain and the hard part is done.

It is now time to configure PI-KVM/NGINX to use your new certificate. NGINX is the service that provides the web services for PI-KVM. 

cd /etc/kvmd/nginx
cat ssl.conf
ssl_protocols TLSv1.3 TLSv1.2 TLSv1.1 TLSv1;
ssl_ciphers "EECDH+AESGCM:EDH+AESGCM:AES256+EECDH:AES256+EDH";
ssl_certificate /etc/kvmd/nginx/ssl/server.crt;
ssl_certificate_key /etc/kvmd/nginx/ssl/server.key;
add_header Strict-Transport-Security "max-age=31536000; includeSubDomains" always;

mv /etc/kvmd/nginx/ssl/server.key /etc/kvmd/nginx/ssl/server.key.old
mv /etc/kvmd/nginx/ssl/server.crt /etc/kvmd/nginx/ssl/server.crt.old
ln -s /etc/letsencrypt/live/pikvm.contoso.com/privkey.pem /etc/kvmd/nginx/ssl/server.key
ln -s /etc/letsencrypt/live/pikvm.contoso.com/fullchain.pem /etc/kvmd/nginx/ssl/server.crt
systemctl restart kvmd-nginx

Now login to your PI-KVM with your favored browser using the Fully Qualified Domain Name and you should see the lock in the URL bar.

You can now put a done stamp on this one.

joecomo67, 2021-08-19:
thank you for that hint - that is very valuable.

I'll try to automate that via system timer and my providers DNS API - are there any best practices for that.
(e.g. ideally I would like to only switch to rw to actually save the new certificates - but certbot likes to write all those
 locks and logs, haven't found a way to turn them off)

sIKE23, 2021-08-19:
In reality certs are certs are certs. I am using a wild card cert for my PIKVM so I generate the cert on a different system then use WINSCP to copy the Wildcard cert into place on the PIKVM. I really do wish that the certs lasted a good bit longer but I guess the commercial boys would throw a fit.

If your DNS provider supports dynamic DNS you should be able to script creating the requested DNS record.

joecomo67, 2021-08-19:
cleanest way would probably to modify the acme script - but I am too lazy for that and stay with certbot.

mdevaev, 2022-03-21:
@djarbz PiKVM os is read-only by default. How did you deal with it?

djarbz, 2022-03-21:
@mdevaev Ahh, yes, thanks for reminding me.
Obviously for the steps as posted we would need to switch into rw mode and ro at the end.

As for my service, I would imagine the following changes to the certbot.service file would be appropriate, I'll find out in ~90 days.
```
[Unit]
Description=Let's Encrypt renewal

[Service]
Type=oneshot
ExecStartPre=/usr/local/bin/rw
ExecStart=/usr/bin/certbot renew --quiet --agree-tos --post-hook "systemctl reload kvmd-nginx.service"
ExecStartPost=/usr/local/bin/ro
```

mdevaev, 2022-03-21:
Got it :D

mdevaev, 2022-03-21:
@djarbz update kvmd to 3.63, it contans small fix for reloading of kvmd-nginx.

mdevaev, 2022-06-24:
I made native services and scripts to work with certificates and secure/atomic updates (that is, so that certificates are not damaged when the power is turned off and do not require RW on the file system).

If someone interested in trying it out, go to [discord](https://discord.gg/bpmXfz5) at #dev and let me know.

djarbz, 2022-06-25:
@mdevaev I have been playing around with your script and I've run into a few issues.

In your `certonly` logic I had to remove the `--webroot` & `--webroot-path` options in order to use my `--dns-cloudflare` plugin.
https://github.com/pikvm/kvmd/blob/f31663b5f0c32ca0f808c34306577f97c96615fc/scripts/kvmd-certbot#L82-L99
I'm not exposing this server to the internet, so I have to utilize the DNS method to generate my certs.

Also, the deploy hook fails.
These 2 lines need to have the full path for the `server.crt/key`.
It also fails because the `kvmd-certbot` user does not have permissions to create the symlinks in /etc/kvmd/nginx/ssl/.
https://github.com/pikvm/kvmd/blob/f31663b5f0c32ca0f808c34306577f97c96615fc/scripts/kvmd-certbot#L96-L97

I'm guessing the instructions will be the same as I detailed above for the renewal process?

djarbz, 2022-06-25:
For reference, this is the command I am running:
```
DOMAIN=pikvm.domain.tld
kvmd-certbot certonly   \
  --dns-cloudflare   \
  --dns-cloudflare-propagation-seconds 60   \
  --dns-cloudflare-credentials /var/lib/kvmd/pst/data/certbot/.cloudflare.auth   \
  --email user@domain.tld   \
  --server https://acme-v02.api.letsencrypt.org/directory   \
  --agree-tos   \
  -n   \
  -d "${DOMAIN}"
```

djarbz, 2022-06-25:
In addition, this does work to enable the certificates, so your `certonly` & `renew` functions will want to reload Nginx on success..

```
ln -sf /var/lib/kvmd/pst/data/certbot/runroot/config/live/${DOMAIN}/fullchain.pem /etc/kvmd/nginx/ssl/server.crt
ln -sf /var/lib/kvmd/pst/data/certbot/runroot/config/live/${DOMAIN}/privkey.pem /etc/kvmd/nginx/ssl/server.key
systemctl reload kvmd-nginx
```

mdevaev, 2022-06-25:
Please install the 3.113 version and try these steps (note the credentials path, it shoud be runroot):
```bash
DOMAIN=pikvm.domain.tld
EMAIL=user@domain.tld
kvmd-certbot certonly   \
  --dns-cloudflare   \
  --dns-cloudflare-propagation-seconds 60   \
  --dns-cloudflare-credentials /var/lib/kvmd/pst/data/certbot/runroot/.cloudflare.auth   \
  --email $EMAIL   \
  --server https://acme-v02.api.letsencrypt.org/directory   \
  --agree-tos   \
  -n   \
  -d $DOMAIN
kvmd-certbot install_nginx $DOMAIN
kvmd-certbot install_vnc $DOMAIN # if you want
# Check the renewal immediately, just for testing
kvmd-certbot renew --force-renewal
```

djarbz, 2022-06-25:
Thanks, this got a bit further.

A few notes, we will need to put the FS in RW mode and can return RO mode after the install_vnc line.

The renewal throws some errors.
```
[root@pikvm ssl]# kvmd-certbot renew --force-renewal
Saving debug log to /tmp/kvmd-certbot/runroot/logs/letsencrypt.log

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Processing /tmp/kvmd-certbot/runroot/config/renewal/pikvm.domain.tld.conf
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Renewing an existing certificate for pikvm.domain.tld

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Congratulations, all renewals succeeded: 
  /tmp/kvmd-certbot/runroot/config/live/pikvm.domain.tld/fullchain.pem (success)
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
--    INFO -- Opening PST session ...
--    INFO -- PST write is allowed: /var/lib/kvmd/pst/data
--    INFO -- Running the process ...
+ rm -rf /var/lib/kvmd/pst/data/certbot/runroot.new
+ cp -a /tmp/kvmd-certbot/runroot /var/lib/kvmd/pst/data/certbot/runroot.new
cp: cannot create directory '/var/lib/kvmd/pst/data/certbot/runroot.new': Permission denied
--    INFO -- Process finished: returncode=1
```

mdevaev, 2022-06-25:
Contact me via Discord plz. I'm online now.

djarbz, 2022-06-25:
@mdevaev and I worked through some things and will update the `kvmd-certbot` utility accordingly.

During our session, this is what we came up with for a process. The RW/RO for the install section may not be needed depending on the decision to handle that internally in the utility.
```
# Prep the environment if you need an auth file for a dns plugin
kvmd-pstrun -- mkdir /var/lib/kvmd/pst/data/certbot/runroot/ && \
  nano /var/lib/kvmd/pst/data/certbot/runroot/.cloudflare.auth && \
  chmod 600 /var/lib/kvmd/pst/data/certbot/runroot/.cloudflare.auth && \
  chown -R kvmd-certbot:kvmd-certbot /var/lib/kvmd/pst/data/certbot

# Issue the initial certificate
DOMAIN=pikvm.domain.tld
EMAIL=user@domain.tld
kvmd-certbot certonly   \
  --dns-cloudflare   \
  --dns-cloudflare-propagation-seconds 60   \
  --dns-cloudflare-credentials /var/lib/kvmd/pst/data/certbot/runroot/.cloudflare.auth   \
  --email "${EMAIL}"   \
  --server https://acme-v02.api.letsencrypt.org/directory   \
  --agree-tos   \
  -n   \
  -d "${DOMAIN}"

# Configure Nginx & VNC to utilize the ACME Certificate
rw
kvmd-certbot install_nginx "${DOMAIN}"
kvmd-certbot install_vnc "${DOMAIN}"
ro

# Check the renewal immediately, just for testing
kvmd-certbot renew --force-renewal
```

**Setup auto renewal**
[Documentation](https://wiki.archlinux.org/title/certbot#:~:text=new%20key%20material.-,Advanced%20Configuration,-Automatic%20renewal)

**Create the renewal service**
`rw
 && nano /etc/systemd/system/certbot.service`

```
[Unit]
Description=Let's Encrypt renewal

[Service]
Type=oneshot
ExecStart=/usr/bin/kvmd-certbot renew
```

**Create the renewal timer**
`nano /etc/systemd/system/certbot.timer`

```
[Unit]
Description=Twice daily renewal of Let's Encrypt's certificates

[Timer]
OnCalendar=0/12:00:00
RandomizedDelaySec=1h
Persistent=true

[Install]
WantedBy=timers.target
```

**Enable the timer**
```
systemctl daemon-reload
systemctl enable --now certbot.timer
ro
```

mdevaev, 2022-06-26:
We have the timer already: `systemctl enable kvmd-certbot.timer`, no need to create it manually.

djarbz, 2022-06-26:
Awesome! Even better!

mdevaev, 2022-06-26:
Thanks for the testing.

https://docs.pikvm.org/letsencrypt

closed 2022-06-26 by mdevaev

realumhelp, 2022-07-01:
> We have the timer already: `systemctl enable kvmd-certbot.timer`, no need to create it manually.

Just got a PiKVM and was running through these instructions, when I try to enable the certbot.timer I get:
```
[root@pikvm ~]# systemctl enable --now kvmd-certbot.timer
Failed to enable unit: Unit file /etc/systemd/system/timers.target.wants/kvmd-certbot.timer does not exist.
```

djarbz, 2022-07-01:
Are you using the latest image with the persistent storage?

realumhelp, 2022-07-01:
I did an update today:

```
[root@pikvm config]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
 there is nothing to do
```

I installed last night and did an update this morning. Not sure if that counts as running the latest or not. I am very new to this project.

djarbz, 2022-07-01:
I think for this to work you need to reflash your SD card with the latest image.

realumhelp, 2022-07-01:
Everything was working except for the kvmd-certbot.timer but I will reload and see what I end up with.

realumhelp, 2022-07-01:
Just freshly downloaded and reinstalled using this image: https://files.pikvm.org/images/v3-hdmi-rpi4-latest.img.xz

Got the same error as above:
```
[root@pikvm ~]# systemctl enable --now kvmd-certbot.timer
Failed to enable unit: Unit file /etc/systemd/system/timers.target.wants/kvmd-certbot.timer does not exist.
```

:shrug:

djarbz, 2022-07-01:
I would suggest opening a new issue so Maxim can take a look at this.

mdevaev, 2022-07-02:
Please check the new image.