pikvm/pikvm #227: Displaying HDMI input to HDMI output instead of a console screen

bozserz, 2021-03-08:
Hello, 
Thank You for the great project. And also thank You to letting me know, that a cheap and good device, ezCoo KVM switch exists. Now I'm trying to use this ezCoo KVM switch to be accessible remotely with the Pi-KVM, everything works properly according to the documentation.
There is just one thing missings: I can connect the local keyboard and mouse to the system, but I cannot connect the local monitor... There are 2 HDMI outputs on a Pi4, but on the first one there is a console screen (not too useful, as console is already available remotely, via SSH or even via the web page), and the second one is not in use.
Is there any command or tool, that can play the stream coming from the HDMI input locally on any of them (without X)? 
It would make it even better.
Regards: Zoltan

arch1mede, 2021-03-08:
Seems like https://github.com/pikvm/pikvm/issues/157, no?

bozserz, 2021-03-08:
> Seems like #157, no?

Well, yes, looks like. Sorry, I did not find that.
Thank You for pointing me there.
Regards: Zoltan.

closed 2021-03-08 by bozserz