pikvm/pikvm #228: Can I use only the web interface of the HTTP protocol?

wnark, 2021-03-08:
How do I turn off https and only use http access, because when I use FRP intranet penetration, an error 502 will be reported when I use the https protocol。
At present, FRP is more compatible with HTTP, so I want to use HTTP only
Thank you

wnark, 2021-03-08:
Sorry, I ignored this question that was asked before, sorry
[close https](https://github.com/pikvm/pikvm/issues/215)

closed 2021-03-08 by wnark

mdevaev, 2021-03-08:
It looks like the problem solved itself :)

wnark, 2021-03-08:
> It looks like the problem solved itself :)

sorry,i still have trouble
i have deleted` return 301 https://$host$request_uri;`
but It reported an error
what should i do？
[image]

mdevaev, 2021-03-08:
You probably missed a line. Look carefully at what exactly you should comment (not only 301): https://github.com/pikvm/kvmd/blob/master/configs/nginx/nginx.conf#L39

wnark, 2021-03-08:
> You probably missed a line. Look carefully at what exactly you should comment (not only 301): https://github.com/pikvm/kvmd/blob/master/configs/nginx/nginx.conf#L39
[image]

I changed it to this, it seems to be useful

mdevaev, 2021-03-08:
Are you saying that it doesn't work like this? Weird. Maybe your browser has cached the 301 forwarding?

wnark, 2021-03-08:
> Are you saying that it doesn't work like this? Weird. Maybe your browser has cached the 301 forwarding?

It should be a cache, http is now normal, thank you