pikvm/pikvm #229: Ability to record on screen movement

arch1mede, 2021-03-08:
Use case:
Capturing targets that lock up or crash, trying to determine why

My specific use case:
Hooking up pikvm to capture the display due to logs not being written as its degrading or locking up

It would be nice to have it start recording as soon as the static screen changs as it starts to dump information to the console/screen.

arch1mede, 2021-03-28:
This actually was supplied by @srepac, he made a script to take a screenshot, compare to another and delete the previous, once reaching a designated threshold, to record 30 seconds. Its actually perfect and a HUGE enhancement to this project.

closed 2021-03-28 by arch1mede

mdevaev, 2021-03-28:
Attach the script plz

srepac, 2021-03-28:
Maxim,

See attached tar file containing three scripts:

  capture.sh
  vid-capture.sh
  monitor-console.sh

Just un-tar into /root/ on pi-kvm and follow instructions in
monitor-console.sh script before running.
Hope this helps

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

andreapx, 2024-09-29:
@srepac I've tried `vid-capure.sh`, it says that it has save the file `/tmp/filename.mp4` (nothing else), but that file doesn't exists.
Also: if a webu session needs to be active, how can I schedule a capture?