pikvm/pikvm #230: Location of main.yaml

Thermi, 2021-03-09:
**Describe the bug**
main.yaml is a vendor/package provided configuration file and thus should be under `/usr/share`, instead of `/etc`. `/etc` is for system specific configuration. `/usr/share` is for vendor/package provided stuff on Arch (and ALARM). A good location would be `/usr/share/pikvm/main.yaml`.
That would also make it evident to any user that it is different from a user specific configuration file.

Thermi, 2021-03-09:
Why do you write this then?
```
NEVER edit /etc/kvmd/main.yaml. Use /etc/kvmd/override.yaml to redefine the system parameters. All other files that are also not recommended for editing have read-only permissions. If you edit any of these files, you will need to manually make changes to them when you upgrade your system. You can view the current configuration and all available KVMD parameters using the command kvmd -m.
```

mdevaev, 2021-03-09:
After editing, of course, he takes on the responsibility of keeping the contents of this file up to date.

Thermi, 2021-03-09:
Why don't you make it so the contents of override.yaml override the read stuff from main.yaml? That way a user can just declare, for example, an empty set, dictionary, or list, ..., for the stuff he/she wants disabled.

mdevaev, 2021-03-09:
This works right now for all types except dictionaries, which are a configuration tree. In any case, someone may want to change the platform settings, and override.use yaml to configure a specific kvm instance.

In fact, there are much more system settings, about two hundred parameters that are provided with a default value. Roughly speaking, this is exactly what could lie in `/usr/share`. That is, the system has default parameters, followed by the platform parameters (main.yaml) and then the parameters of a specific instance (override.yaml).

Even if this separation seems artificial, it is important to keep the configuration consistent. For example, nginx is configured on the same principle (`/etc/kvmd/nginx`). And even if I agreed that the configuration does not belong there (and I do not agree for the reasons mentioned above), from the point of view of backward compatibility and user experience, it is a VERY bad idea to change the paths and roles of configuration files.

Thermi, 2021-03-09:
I see.

closed 2021-03-09 by Thermi