pikvm/pikvm #232: Pi Zero W image has no keyboard support

cshabazian, 2021-03-13:
The current v2-hdmi-zerow.img doesn't allow a keyboard to work.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Create image and boot Pi Zero W
2. Try to access the Pi Zero W with a keyboard (tried both physical and a pikvm running on a Pi 4)
3. Keyboard doesn't work
4. Rebooting the same Pi Zero W with a stock Pi image works as expected

**Expected behavior**
The keyboard works

**Pi-KVM info:**
  - Raspberry Pi board version: Pi Zero W
  - Pi-KVM platform:  v2-hdmi-zerow

mdevaev, 2021-03-13:
This is not a bug. The USB controller can only operate in one of two mutually exclusive modes: host mode and device mode. The image uses device mode (to emulate the keyboard, disk, and everything else), so you can't connect another device to it.

cshabazian, 2021-03-13:
Then how do you configure the wifi?

mdevaev, 2021-03-13:
https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md#the-final-steps

closed 2021-03-13 by mdevaev