pikvm/pikvm #236: No picture on PIKVM with Displayport to Triple Monitor Output

JsBergbau, 2021-03-17:
**Describe the bug**
I don't know if this is really an issue, however perhaps you can help.
I bought this https://www.amazon.de/gp/product/B00V07XW3E device. It outputs from Displayport to VGA, DVI and HDMI. I use DVI for connection to monitor and want to use HDMI for connection to PI-KVM. However I always get "No signal" on PIKVM. If I plug this HDMI cable to another monitor I get the same picture on both monitors. If I plug a Displayport to HDMI adapter to PIKVM I get the signal on the PI-KVM.

Any idea whats going on here? 
**To Reproduce**
You need this special displayport adapter, see above.

**Expected behavior**
With that display port adapter I see the picture on the monitor and on PI KVM.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**Pi-KVM info:**
  - Raspberry Pi board version [e.g. RPi 4]: PI Zero
  - Pi-KVM platform [e.g. v2-hdmi] v2-hdmi
  - Video capture type [e.g. CSI bridge] CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
Add any other context about the problem here.

mdevaev, 2021-03-20:
To be honest I have no idea. Some adapters is working, and some not. I have one displayport adapter and I use it constantly with pikvm, everything works. But I think that this will not help you much.

JsBergbau, 2021-03-20:
Which model do you use?

mdevaev, 2021-03-20:
Noname. Looks like this:
[image]

JsBergbau, 2021-03-20:
Oh Ok. I've used different single Displayport to HDMI adapters. All these worked. It came only to problems with HDMI 1 to 2 Splitter or that above linked device.
Currently I use this one https://www.amazon.de/DeLock-87720-Displayport-Splitter-schwarz/dp/B079Z89CV6
This works fine. On Windows it enables me even to use 4 different displays with AMD Radeon Ryzen 3 Pro 4350GE with integrated Radeon Graphics and on Boot it shows the same image like the monitor which enables it to enter Bitlocker Password via PI-KVM. This is my only usecase for PIKVM because once I'm in Windows I can use RDP to connect. 
However this particular Delock device has one major disadvantage: It needs to be USB powered and there it consumes almost always 2 Watt, even when there is no monitor signal. I've ordered another Splitter and will report when I've tested it here.

JsBergbau, 2021-03-23:
Now I've ordered https://www.amazon.de/PremiumCord-DisplayPort-Monitore-Extended-Duplicate/dp/B08438TV6G/
Needs to be connected via Micro-USB cable to get power, however there is no powerdraw. 
When monitor is off it still consumes 3W from DisplayPort.
In Bitlocker resolution is usually 1280x720. With that resolution images is blinking. So I can't recommend with PIKVM.
With AMD Radeon Ryzen 3 Pro 4350GE with integrated Radeon Graphics are 4 monitors possible. A 5th monitor is not supported. If I pull the plug of another monitor I can use all 3 HDMI outputs of the Displayport adapter.

mdevaev, 2021-07-14:
I think there is nothing to do here now, but this information is valuable for other users. Thanks!

closed 2021-07-14 by mdevaev

mdevaev, 2026-01-02:
@ebarrere There is no list now, but if you want to share something, you could create a new issue and we start such list.

PS: Happy New Year!