pikvm/pikvm #240: Screen size bigger than browser stream widget

rmatinata, 2021-03-23:
**Describe the bug**
I have a Pi zero W, connected to a PWAY HDMI KVM switch through a HDMI-to-CSI2, which in turn is connected to 4 desktop PCs (all of them with DP-to-HDMI cables). Everything is working fine (apparently), however it seems that the actual screen size is bigger than what the Browser stream widget is showing. The effect is that probably 40% of the screen lines a hidden. I can, however, see the screen scrolling, so if you hit enough "enters" in a Linux console, one would see the content scrolling up.

I honestly cannot relate this behavior to any of the suggestions around, so what could be tweaked in pikvm so the actual HDMI stream screen size matches what the browser stream widget shows ?

By the way, the stream size option in the UI only changes the browser stream widget size, but not the actual HDMI stream resolution. So in effect, the widget is bigger, the characters are bigger, but the actual HDMI stream resolution is unchanged (still 40% of the screen lines are hidden).

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**Pi-KVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
  - Pi-KVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
Add any other context about the problem here.

rmatinata, 2021-03-23:
I just confirmed that VNC shows the same behavior as the browser.... If i resize the TigetVNC window, the actual screen size does not change (obviously).

mdevaev, 2021-03-23:
Probably your OS gives such a picture, you need to look at the parameters of the resize on the computer side.

You can't change the resolution on the pikvm side, as it behaves like a monitor. You can't tell the monitor what resolution the OS should use.

mdevaev, 2021-04-01:
I think I answered your question

closed 2021-04-01 by mdevaev