pikvm/pikvm #245: Changing some /usr/bin instances to /bin, /usr/sbin, etc would help non-arch users

aggieNick02, 2021-03-26:
If you're open to this, I'm happy to put together a quick PR for it. On arch /usr/bin, /usr/sbin, and /bin are the same thing. But not on Raspberry PI OS among other things. This causes some commands like `ip_cmd`, `iptables_cmd`, etc to require entries in override.yaml to get things going on alternative setups, since they point to /usr/bin but the tools live in /bin for certain other OS users.

The ones I've found so far - 
ip_cmd - /usr/bin/ip => /bin/ip
iptables_cmd - /usr/bin/iptables => /usr/sbin/iptables
pre_stop_cmd - /usr/bin/systemctl => /bin/systemctl
post_start_cmd (argument) - /usr/bin/dnsmasq => /usr/sbin/dnsmasq

Related, I'm guessing if someone fully installs the kvmd,etc apps, those prefixes would be /usr/sbin instead of /usr/bin on many OSes. I'm currently using kvmd by invoking via python instead of running an installed app, so it doesn't affect me, but it's just something I noticed. If you'd like me to look at changing those up as well, I'd be happy to.

mdevaev, 2021-04-01:
I'll think. I usually don't change the default values of the parameters for backward compatibility reasons. If it doesn't break anything, then maybe I will.

mdevaev, 2021-07-14:
Okay, I'll do it, but not in the first place. I will install Raspberry OS to see what paths are used there to ensure 100% compatibility.

mdevaev, 2021-08-11:
It seems that in Raspberry OS some directories already refer to /usr. All I need is to process /sbin.

```
root@reki:/# ls -l /
total 56
lrwxrwxrwx   1 root root     7 May  7 15:39 bin -> usr/bin
drwxr-xr-x   3 root root  3584 Jan  1  1970 boot
drwxr-xr-x  22 root root  4000 Aug 11 13:35 dev
drwxr-xr-x  80 root root  4096 Aug 11 14:22 etc
drwxr-xr-x   3 root root  4096 May  7 15:42 home
lrwxrwxrwx   1 root root     7 May  7 15:39 lib -> usr/lib
drwx------   2 root root 16384 May  7 15:59 lost+found
drwxr-xr-x   2 root root  4096 May  7 15:39 media
drwxr-xr-x   2 root root  4096 May  7 15:39 mnt
drwxr-xr-x   3 root root  4096 May  7 15:42 opt
dr-xr-xr-x 302 root root     0 Aug 11 13:36 proc
drwx------   2 root root  4096 May  7 15:39 root
drwxr-xr-x  29 root root   740 Aug 11 14:25 run
lrwxrwxrwx   1 root root     8 May  7 15:39 sbin -> usr/sbin
drwxr-xr-x   2 root root  4096 May  7 15:39 srv
dr-xr-xr-x  13 root root     0 Aug 10 20:32 sys
drwxrwxrwt   2 root root    40 Aug 11 14:26 tmp
drwxr-xr-x  10 root root  4096 May  7 15:39 usr
drwxr-xr-x  12 root root  4096 Aug 11 14:21 var
root@reki:/# cat /etc/rpi-issue
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2
```

closed 2021-08-11 by mdevaev