pikvm/pikvm #246: Linux fails to boot with remote power / reset connected

anrstone, 2021-03-29:
**Describe the bug**
I added the remote power on / reset / LED functionality and the LEDs work as expected as does the power on / reset. But with the remote buttons connected to the machine Linux immediately shuts down after the initial linux boot menu

**To Reproduce**
Steps to reproduce the behaviour, like:
1. Press short or long remote power on
2. Wait for the Linux boot menu to appear
3. Either allow the auto selection to take place or manually select the boot image
4. The machine powers off

**Expected behavior**
I would expect with the remote power on that the machine should boot normally

**Desktop (please complete the following information):**
 - OS: Linux CNC (Debian Buster)
 - Chrome

**Additional context**
If I remove the remote control and allow the machine to boot normally then reconnect the remote the machine works fine which suggests to me that perhaps either the power or reset are still enabled after initial use or that the kernel is picking something up from the button positions. I have tried the remote reset at the initial linux menu screen and then tried to let the machine boot normally but no improvement.

mdevaev, 2021-03-29:
You probably assembled the wrong circuit and / or used the wrong optocouplers.

anrstone, 2021-03-31:
Turns out it was a faulty motherboard. If you put a scope on the PWR button without any kvm connected and use a manual push button you can see the voltage dip just after the linux boot screen which causes the machine to power down. I swapped the PSU with a few others and no joy so I'm guessing there's either a faulty power connection on the board or faulty connector to the switches. I have adapted your original circuit to hold up the voltage on the PWR button line which works fine...

closed 2021-03-31 by anrstone