pikvm/pikvm #251: Docker image

sobake, 2021-03-31:
**Is your feature request related to a problem? Please describe.**
I'd like to run multiple container applications on my PI 4.

**Describe the solution you'd like**
A docker image for pikvm.

zsmbrvr, 2021-04-02:
Might be possible for USB, but I am not sure if you can access the the CSI bridge in docker.

closed 2021-04-02 by sobake

mdevaev, 2021-04-02:
[KVMD](https://github.com/pikvm/kvmd) and other services run inside Docker for debugging. Docker + QEMU is also used to [build images](https://github.com/pikvm/pi-builder). I believe that it is possible to combine both of these approaches. But alas, I have more priority tasks.

tomaszduda23, 2021-11-15:
@mdevaev PoC in docker is quite easy. The implementation assume that systemd, dbus which is great for OS. In case of docker seems a little bit complex. Would you be willing to accept patches to make it lightweight or configurable to run in docker?

mdevaev, 2021-11-15:
Only if it won't change the default behavior