pikvm/pikvm #252: Update option in web GUI

sharicov, 2021-04-01:
Could you please add update button in the web GUI.
Update is the only reason I connect to pikvm in terminal, would be much more convenient to do a system update with single click instead.
Thanks

zsmbrvr, 2021-04-01:
I agree, it would be nice, but would it only update pikvm or the whole system?

mdevaev, 2021-04-01:
Pikvm is updated using the package manager along with the rest of the operating system. The update must be performed entirely to avoid library conflicts and incompatibilities. I don't think it makes sense to make a special button for this in the web ui, since there is a great web terminal where updates are made using a couple of commands.

What if something goes wrong during the update? You need to save the log, but how will you handle the error? The maximum that we need to do here is to write a small script that combines several update actions into one.

sharicov, 2021-04-02:
Most of the times update should work well, for rear occasions that it doesn't error message should be sufficient (display only). So button would make no problems

mdevaev, 2021-07-16:
Perhaps it will be done in the future. Right now it's a duplicate of #47

closed 2021-07-16 by mdevaev