pikvm/pikvm #259: Docker image?

sobake, 2021-04-02:
**Is your feature request related to a problem? Please describe.**
Looking to implement a docker image for pikvm so Pi 4 can be used for more than one web server application.

**Describe the solution you'd like**
From what I understand, docker can passthrough usb ports from the host. Can the rest of the implementation pikvm uses also be run from a docker container without access to the host kernel?

mdevaev, 2021-04-02:
See my answer on #251

mdevaev, 2021-05-14:
This will not be done for the foreseeable future, sorry. Someone who needs it can do. Too much time for implementation, despite the fact that only a few will use it. The official OS image does not require this.

closed 2021-05-14 by mdevaev