pikvm/pikvm #260: override.yaml for serial: drives: and ethernet: under otg: causes all these to not work and cannot reboot pi

srepac, 2021-04-03:
BACKGROUND:

Originally, /etc/kvmd/override.yaml only had this:

otg:
    devices:
        drives:
            enabled: true  # Set it to true to enable
            count: 2  # +1 drive, default value
            default:  # Default configuration for the all extra drives
                cdrom: false  # Default value (false for the generic flash drive)
                rw: true # Read-only by default
# enable serial console to pi-kvm from managed server
        serial:
            enabled: true

Serial console and USB flash mount working as expected on connected server.

After /etc/kvmd/override.yaml updated with the following, then rebooted pi:

# USB ethernet passthrough to server
        ethernet:
            enabled: true
            driver: rndis
            host_mac: 48:6f:73:74:35:43
            kvm_mac: 42:61:64:55:35:42
otgnet:
    firewall:
        allow_tcp: [80, 443]
        forward_iface: wlan0
    commands:
        post_start_cmd_append:
        - "--dhcp-option=6,1.1.1.1"
    iface:
        ip_cmd:
            - /usr/bin/ip
        net: 10.35.0.0/28

All of these OTG devices (serial, drives, and ethernet passthru) fail to work.

SYMPTOM:

- USB flash mount is not working after override update and reboot; mounted in terminal and verified it's there, but win host doesn't see it
- Serial console is not working (no COM port on win host presented by pi)
- Rebooting the pi will just get stuck at "Stopped Getty on ttyGS0." and never reboots.  Have to remove/reconnect power for pi to boot.

ISSUE (based on journalctl -xe):

The issue seems to be that because there is already a usb_gadget being used, OTG setup never starts.  So, you can't have all these combinations in the override.  If I remove the ethernet: override, serial/usb flash mount work.

Apr 03 08:50:12 pimsd systemd[1]: Starting Pi-KVM - OTG setup...
Apr 03 08:50:19 pimsd kvmd-otg[1528]: kvmd.apps.otg                     INFO --- Using UDC 20980000.usb
Apr 03 08:50:19 pimsd kvmd-otg[1528]: kvmd.apps.otg                     INFO --- Creating gadget 'kvmd' ...
Apr 03 08:50:19 pimsd kvmd-otg[1528]: kvmd.apps.otg                     INFO --- MKDIR --- /sys/kernel/config/usb_gadget/kvmd
Apr 03 08:50:19 pimsd kvmd-otg[1528]: Traceback (most recent call last):
Apr 03 08:50:19 pimsd kvmd-otg[1528]:   File "/usr/bin/kvmd-otg", line 9, in <module>
Apr 03 08:50:19 pimsd kvmd-otg[1528]:     main()
Apr 03 08:50:19 pimsd kvmd-otg[1528]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/otg/__init__.py", line 289, in main
Apr 03 08:50:19 pimsd kvmd-otg[1528]:     options.cmd(config)
Apr 03 08:50:19 pimsd kvmd-otg[1528]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/otg/__init__.py", line 176, in _cmd_start
Apr 03 08:50:19 pimsd kvmd-otg[1528]:     _mkdir(gadget_path)
Apr 03 08:50:19 pimsd kvmd-otg[1528]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/otg/__init__.py", line 52, in _mkdir
Apr 03 08:50:19 pimsd kvmd-otg[1528]:     os.mkdir(path)
Apr 03 08:50:19 pimsd kvmd-otg[1528]: FileExistsError: [Errno 17] File exists: '/sys/kernel/config/usb_gadget/kvmd'
Apr 03 08:50:19 pimsd systemd[1]: kvmd-otg.service: Main process exited, code=exited, status=1/FAILURE
Apr 03 08:50:19 pimsd systemd[1]: kvmd-otg.service: Failed with result 'exit-code'.
Apr 03 08:50:19 pimsd systemd[1]: Failed to start Pi-KVM - OTG setup.

This is the second time this has happened, so it is repeatable.

srepac, 2021-04-03:
Based on my testing, OTG combinations that work:

- serial and usb
- ethernet and usb

OTG combinations that don't work:

- serial and ethernet   or any combinations involving these two in the mix (i.e. the reason I opened this issue)

As of now, only use **one** or the **other**, **NOT both**.  If you do, you break all OTG functionality altogether.

mdevaev, 2021-04-04:
Apparently this is a kernel bug and should be fixed by the developers.

mdevaev, 2021-07-16:
I will look at what is happening there with the protocol analyzer, maybe I will see something interesting.

mdevaev, 2021-07-29:
I see the bug but don't understand it yet.

mdevaev, 2022-10-28:
https://github.com/RoganDawes/P4wnP1_aloa/issues/131

mdevaev, 2023-03-26:
It's not a conflict but the hardware limitation of Raspberry Pi. I've documented it on each page related with OTG.

closed 2023-03-26 by mdevaev