pikvm/pikvm #263: Multiple Monitor / Dual HDMI

esticle, 2021-04-07:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Support for multiple display inputs would be fantastic

This was once referenced as a feature to come in the original announcement [Reddit post here](https://www.reddit.com/r/sysadmin/comments/imrzyp/pikvm_raspberrybased_advanced_diy_kvm_over_ip/g4hfsbp?utm_source=share&utm_medium=web2x&context=3)

ie. 2x HDMI inputs - this will take care of some docked laptops, etc.

PS - raising as I have not seen such a official feature request

**Describe alternatives you've considered**
None available

**Additional context**
N/A

wnark, 2021-04-07:
You can use kvm switch to achieve this function

esticle, 2021-04-07:
> You can use kvm switch to achieve this function

How so? A kvm switch will switch between inputs and would not output two simultaneous displays to one output.

I would like to take a dual display input (thus the 2xHDMI support ask) and have pikvm show it in one browser/vnc window.

wnark, 2021-04-07:
`https://detail.tmall.com/item.htm?spm=a1z10.1-b-s.w5003-22991446579.1.736834b4znTpg4&id=613028097354&rn=8d4452a7e26cdb682c4f0974f8f09ade&abbucket=12&scene=taobao_shop`
Or try this thing？

arch1mede, 2021-04-08:
https://github.com/pikvm/pikvm/blob/master/pages/Community_FAQ.md#can-you-have-the-pi-kvmrpi4-connected-along-with-a-monitor , look near the bottom of that...hdmi loop will do what you are looking for

esticle, 2021-04-09:
H,

> `https://detail.tmall.com/item.htm?spm=a1z10.1-b-s.w5003-22991446579.1.736834b4znTpg4&id=613028097354&rn=8d4452a7e26cdb682c4f0974f8f09ade&abbucket=12&scene=taobao_shop`
> Or try this thing？

@wnark Yep there are a couple commercial solutions but you're looking at €600+ for the better ones with VNC support.

> https://github.com/pikvm/pikvm/blob/master/pages/Community_FAQ.md#can-you-have-the-pi-kvmrpi4-connected-along-with-a-monitor , look near the bottom of that...hdmi loop will do what you are looking for

@arch1mede That would be for cloning/simultaneous use of a existing display, not encoding 2 display outputs  - not what I'm after.

If it was not clear - Imagine I want to pikvm a RPI4/any desktop utilising both HDMI outputs that would typically go to two monitors but want to IP-KVM it with something like pikvm instead.

The only way (I can think of) to achieve that at a cheaper-than-commerical pricepoint is to get 2xpikvm's and simply align two browsers on a larger display close to each other - however since the author originally commented on a new version possibly having another HDMI input it would/should achieve the same thing (I assume having a second camera module is the actual issue at hand).

arch1mede, 2021-04-16:
@esticle Its prob not possible till the Pi has enough bus BW to allow multiple ingress/egress streams.

mdevaev, 2021-07-14:
Related: #236

mdevaev, 2022-08-01:
Pi does not have the resources to encode two sources at the same time, so it will can't be done in the form described.

closed 2022-08-01 by mdevaev