pikvm/pikvm #270: Device Data for auto configure and send data back to let the community know of used hardware

Kaydron1000, 2021-04-12:
**Is your feature request related to a problem? Please describe.**
I see multiple posts looking for which USB dongle to buy or what HDMI-CSI to buy (there might be 2 or 3 different ones out there). Along with that there is several places in the docs stating to change config here and there to get desired features for particular hardware. I also saw a post about overheating issue. Temperature data peak/avg could be sent back to developer too.

**Describe the solution you'd like**
With these 2 issues something similar can resolve both of them:
- Grab hardware info and send back to developer to assist design and let the community know the popular or working models.
- Grab hardware info to assist with shell scripts menu to automatically set configs for the selected options in menus.

I was looking into smxi.org/inxi, lsscsi, or any of those tools  to pull info out from your hardware. 

**Describe alternatives you've considered**
Alternative is skip the automated part of scanning hardware and just give options to the user.

**Additional context**
none

arch1mede, 2021-04-13:
@Kaydron1000 
The community faq has a list of names used for the CSI modules
I can submit a PR on the USB device that works for me
Heating should be handled by the end user
I don't know what you are going after on the auto set config verbiage, I will let the dev speak to this

Kaydron1000, 2021-04-14:
There was 2 items I was trying to knock out there:
1. I saw issue #182 that a couple Pi ZeroW overheated somehow? probably soldering mishap. I was thinking sending out info of connected hardware, peak temp and avg temp each day would assist in troubleshooting and developing software. I actually am siding against this more so since it is unknown if the user shoved their device in a stuffed animal or not (almost unless data then). Also I am not the biggest fan of sending info back out since usually it is masked what is actually being sent out in most cases and with the previous statement not a big need.
2. The idea I had for auto configure if the different images get merged (usb vs. csi) would be one instance to check hardware then set things accordingly (i guess essentially like plug n play??) that requires a lot more dev work. The more in the near future is setting up a shell file with menu options of setting up different config items. I have not dug into these setups yet, but all these seem to be setting the override file and running some commands:
- IPMI BMC
- IPMI SoL
- Redfish
- Bluetooth HID
- Ethernet over USB
- Moues Modes (relative vs abs)
- Tailscale VPN
- VNC
- Wifi setup

There exist some config that is difficult to do through this method since it depends on what is connected to your GPIO pins which ends up being more advance setup.

This would provide a more user friendly interface for users to configure their setup. Likewise these options could be available at the client web interface and commands could be passed to this shell script. Which would make it even more user friendly.

arch1mede, 2021-06-10:
My 2 cents, I would close this issue, not even sure this is an issue or a valid problem/request. The dev has stated he will not program this project to phone home, ever.

mdevaev, 2021-07-16:
The problem of dead Zero may well be related to a chip defect (I have already encountered a similar one). And these are the only devices that have died because of this. Even if I were collecting statistics, it should be impersonal, and this would not help me in any way to understand that these devices broke down due to overheating (even if this is the case).

Merging several images into one is impossible at least because of the different CPU architecture (for RPi4 and Zero). And I really don't think that choosing one image from the list is a big problem.

And most importantly, I really, really don't like the idea of collecting any statistics for a device that can be used for something that other people shouldn't know about (trade secrets, etc). As soon as telemetry appears in the software, you can no longer trust it.  I would prefer to keep the project's reputation clean and not collect telemetry at all.

As for the configuration scripts, this is a good idea that can be implemented in task #47. But now I have a lot of other more priority tasks.

closed 2021-07-16 by mdevaev