pikvm/pikvm #277: Add timezone to readme.md

iaredavid01, 2021-04-13:
Possibly under tips section, add instructions for setting system time zone via SSH:
`timedatectl set-timezone America/New_York`
possibly with a link to [the tz list on wikipedia](https://en.wikipedia.org/wiki/List_of_tz_database_time_zones).

I changed mine for logging purposes, but I imagine it would be useful to novice users (based on the kind of questions I see on the subreddit).

srepac, 2021-04-13:
This is already covered in the community_faq.md 

https://github.com/pikvm/pikvm/blob/master/pages/Community_FAQ.md#things-to-do-after-initial-install

iaredavid01, 2021-04-13:
So strange that I didn't catch that... huh. Thanks for the correction

closed 2021-04-13 by iaredavid01