pikvm/pikvm #280: 500 Internal Server Error after pacman update

DerSeb90, 2021-04-16:
**Describe the bug**
Fresh Install of PiKVM from Image. The boot and ssh into. Then like advised doing an Update with:
rw
pacman -Syy
pacman -Su
reboot
After the Reboot the Webinterface shows just 500 Internal Server Error
During install is a Warning with Incompatibility with python and python-pikvm

**Expected behavior**
A working webinterface.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: MS Edge

**Pi-KVM info:**
  - Raspberry Pi board version RPi 4
  - Pi-KVM platform [e.g. v2-usb]

mdevaev, 2021-04-16:
Already fixed. Use this:
```
pacman -Syy
yes | pacman -Sdd python python-aiohttp
pacman -Su
reboot
```

DerSeb90, 2021-04-16:
Thanks worked!
Just 1 question. Isn't there a button in the webinterface to reboot/poweroff the whole pi? The Web Terminal gives an error on the command reboot.
So it's just possible via ssh?

mdevaev, 2021-04-16:
Use webterminal:
```
su -
<enter root password>
poweroff
```

DerSeb90, 2021-04-16:
Ok thanks. Problem super fast solved!

closed 2021-04-16 by DerSeb90

mdevaev, 2021-04-16:
Welcome!

colindawson, 2021-08-28:
I'm experiencing this problem.  Looks like it's not resolved.

rw
pacman -Syy
pacman -Su
reboot

for the 500 server error.   Then I found this page, and executed 

pacman -Syy
yes | pacman -Sdd python python-aiohttp
pacman -Su
reboot

however, I'm still getting the 500 server error.