pikvm/pikvm #281: relative mouse - scale sensitivity according to viewport

benbarkay, 2021-04-18:
Using pikvm with relative mouse in a scaled viewport causes the mouse to move too fast if the viewport is bigger than the source, or too slow if the viewport is smaller than the source.

mdevaev, 2021-05-16:
@benbarkay boop

benbarkay, 2021-05-16:
@mdevaev Sorry, did not receive notification for your first reply.

I use Chromium. The viewport of the "server" is 720p (Android), and I run the browser in a 4k monitor.
Let me know if I can be of any more help

mdevaev, 2021-07-14:
I'm just letting you know that I haven't forgotten about it. I'll get a high-resolution screen and try to figure it out. Or at least I will make a slider for the sensitivity of the mouse.

benbarkay, 2021-07-24:
@mdevaev that would be really great.
Please let me know if I can be of any help.

mdevaev, 2021-08-17:
Done. Will be available in the next release.

closed 2021-08-17 by mdevaev