pikvm/pikvm #282: otgnet - add bridge option

benbarkay, 2021-04-18:
Exposing the server to the network that pikvm is connected to with `forward_iface` is difficult, I would suspect that many users would have liked the server to obtain an IP from the upstream router that pikvm is connected to.

Adding a configuration option to set-up a bridge would be convenient. I am currently setting this up by disabling `kvmd-otgnet` and using the following netctl profile:
```
Description="otgnet bridge"
Interface=br0
Connection=bridge
BindsToInterfaces=(usb0 eth0)
MACAddress=eth0
IP=dhcp
## Ignore (R)STP and immediately activate the bridge
SkipForwardingDelay=yes
```

This results in that both pikvm as well as the server receive an IP from the upstream router's dhcp server, as though both are connected to the upstream router directly.

srepac, 2021-04-19:
@benbarkay I've created a bridge-net.sh script exactly for this purpose.   https://pastebin.com/ZrurJL4X

benbarkay, 2021-04-19:
> It is assumed that between the pikvm and the server should operate its own small internal network for service needs, which does not intersect with the main network. The forward_iface option is needed for rare cases when you want pikvm to be a router for some reason. Bridge is an even more specialized case, I think

I think that while pikvm does cover the uses where a "normal" kvm would have been appropriate, it additionally covers use cases where a "normal" kvm wouldn't be sufficient or appropriate.
This is one of these features that might not make a lot of sense in a server environment, but would be enormously useful in others (such as my type of use). 

I just think it would have been cool if this was streamlined into otgnet's configuration, or maybe even making it to the UI at some point. Or maybe even just adding this to the cookbook would also be sufficient.

You can close this issue if that's not very convincing to you :smile:

mdevaev, 2021-04-26:
Well, convincing or not, but you need it, so there is such a usecase. I can't promise that I'll do it quickly, but I won't close this issue. When I get my hands on it, I'll deal with it.

arch1mede, 2021-07-06:
@mdevaev , doesn't @srepac 's script actually do what he is asking for or are you wanting this added to the gui?

mdevaev, 2021-07-14:
@arch1mede I don't remember

mdevaev, 2021-07-16:
@arch1mede so, I'm stupid. I will make this functionality in kvmd-otgnet.

cipy, 2022-02-02:
well, I just tested @srepac and

- something works (server does get an IP via the outside router)
- then nothing works, no ping, nothing seem to reach the server again
- the bridge seems to be busy with some traffic coming over the outside network 
- but the PiKVM interface (long name) towards the server seems to be pretty quiet 😾

srepac, 2022-02-02:
What exactly did you do?  Did you add usb ethernet adapter to pikvm?  Did you run my bridge-net script?  Was the pikvm set to use DHCP IP on its ethernet?

cipy, 2022-02-03:
@srepac 

I use the default config of PiKVM => USB cable + USB ethernet dongle => server
yes, the PiKVM eth0 did have a DHCP running on it as in its default config
the server's eth0 did get an IP address over DHCP from the (same) external router

the server did transmit over its eth0 (tcpdump)
the PiKVM did not see anything on the (long interface name of the) USB dongle
the br10 on PiKVM was busy doing things (tcpdump, so it did get to join PiKVM's eth0 and USB dongle)

I flushed the firewall (when OTG was active with network over the USB cable) and still ⛔ traffic at all

when I rebooted at some point the PiKVM I did see 2 pings being answered on the server's console 📦

extrememicro, 2022-09-24:
@cipy @srepac 
I don't understand why bother using an usb ethernet dongle. I only use USB-C emulation port to share the same pikvm lan to the server.

To config bridge instead of routing, edit /etc/netctl/kvm-bridge like @benbarkay mentioned

```
Description="Bridge Interface br10: eth0 usb0"
Interface=br10
Connection=bridge
BindsToInterfaces=(eth0 usb0)
MACAddress=eth0
IP=dhcp
# If you want also for DHCPv6,uncomment below line
#IP6=dhcp
## Ignore (R)STP and immediately activate the bridge
SkipForwardingDelay=yes
```
Then use the commands: 
```
netctl enable
echo "net.ipv4.ip_forward = 1" > /etc/sysctl.d/99-kvmd-extra.conf
```

Add to /etc/kvmd/override.yaml like mentioned in docs but forwarding bridged interface:

```
otg:
    devices:
        ethernet:
            enabled: true
            driver: rndis
            host_mac: 48:6f:73:74:50:13
            kvm_mac: 42:61:64:55:53:12
                
otgnet:         
    firewall:
        forward_iface: br10      
```      

Finally reboot.

Stitch10925, 2023-01-13:
Sadly enough this approach doesn't work to bridge wlan0. It doesn't allow you to bind to the wlan0 interface.

BigRedAwesomeDude, 2023-10-26:
Thank you for this, and I'd like to throw my 2 cents in for getting this added as a standard part of PiKVM without all the mess.

My server uses bonded 10GbE on a vLAN trunk for its primary networking.  I set up the bridge via PiKVM as a backup network connection for management in the event something goes pear-shaped with the primary network setup.  The whole point of PiKVM is for backup connectivity when things don't work as planned.  Now I also have a backup network connection without burning another port on my switch.  I have limited limited copper 1Gbe ports so I need to use them wisely.