pikvm/pikvm #283: Updated layout for documentation pages

Kaydron1000, 2021-04-19:
**Is your feature request related to a problem? Please describe.**
The current large README doc has a lot of information jumbled around and several markdown files in the pages folder that are hidden unless someone clicks through-out the whole pikvm repository.

**Describe the solution you'd like**
In short updated documentation utilizing GitHub project website. I have a presentable version here: https://kaydron1000.github.io/pikvm/

**Describe alternatives you've considered**
Alternatively a wiki could be made up but I have not used GitHub wiki before.

**Additional context**
I went through the large README.md and split up into individual pages accordingly and included all the markdown files in the pages folder. Please review and let me know about the content at the website (https://kaydron1000.github.io/pikvm/). I will be updating the layout to something slightly different. https://squidfunk.github.io/mkdocs-material/getting-started/. I am trying to get this finished before the v3 goes on sale.

After I get mkdocs layout working we can move over to this repository and manage and update from here.

White-SAndS, 2021-04-19:
The "Building OS" Page is somewhat misleading. there is statet that arch is working for building v0 for different devices but shown commands are not working on arch. arch has no apt-get. it looks more like debian?!?

Kaydron1000, 2021-04-19:
Those are the original instructions. I do agree it is misleading. I'm not well versed on Linux distributions but I am familiar with using apt-get on Raspbian / ubuntu / Debian. I thought Arch used pacman.
I'll add at the top ### Following Instructions can be used on a Debian distribution

arch1mede, 2021-06-10:
My 2 cents, I would close this issue. IF any document is to be accepted as a replacement to what is already offered, then it should be by someone who is a more active user and actually have the hardware and update it consistently. This github.io hasn't been updated in 2 months and the OP hasn't been seen in the same time frame, prior to that it was the same sporadic behavior.

mdevaev, 2021-07-24:
I really like the work done and I definitely use templates and markup to create a website with documentation in the future. So I'll keep this issue. Thank you for that @Kaydron1000