pikvm/pikvm #286: PowerShell windows with colored output interfere with rendering

strausmann, 2021-04-23:
**Describe the bug**
When outputting text in color in a PowerShel window, the entire image starts to flicker. Here a Chocolatey update was executed for all installed applications. As soon as the PowerShell window is minimized, the rendering of the desktop is possible again without errors. Apparently only certain output in PowerShell is responsible for the problem.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connected to Dell Latitude Laptop with Windows 10
2. Open a PowerShell Window and update all Applications with Chocolatey "choco update all -y"
3. Now the rendering has problems with the powershell output

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

https://strausmanncloud-my.sharepoint.com/:i:/g/personal/bjoern_strausmann_net/EXcu87e9ooNLjiowmgd0xw0BXDTGV7j6hQBLjvW7w1ghzw?e=xkkscS

**Desktop (please complete the following information):**
 - OS: Windows 10 64bit 2020
 - Browser chrome, edge chromium

**Pi-KVM info:**
  - Raspberry Pi board version
(RPi4 4 GB)
  - Pi-KVM platform
v2 with hdmi USB dongle
  - Video capture type
hdmi USB dongle

  - KVMD version: `pacman -Q | grep kvmd`

kvmd 2.54-1
kvmd-fan 0.1-1
kvmd-oled 0.3-1
kvmd-platform-v2-hdmiusb-rpi4 2.54-1
kvmd-webterm 0.36-1

  - uStreamer version: `pacman -Q | grep ustreamer`
ustreamer 3.26-1

  - Linux kernel: `uname -a`
Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

strausmann, 2021-04-23:
I think the powershell output in yellow is the problem.

mdevaev, 2021-04-23:
I tend to think that this is just another usb dongle hardware glitch (out of many), that's why I don't recommend using it. It encodes jpeg on its own, so it can't be a problem with software that just transmits what is encoded by dongle.

arch1mede, 2021-04-25:
I would like to add additional context to this issue. I think its worth looking into rather than just putting the blame on a dongle.

Running FF my win10 session with the usb dongle, after bringing up something white, would NOT release the blue mouse dot reference as if the window was locked up or causing a process to hang. 

```Error code: STATUS_BREAKPOINT```
```[2021-04-25 17:29:06 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Got SIGUSR2, checking the stream state ...```
Would just stall the FF window at this point

Switched to Chrome and restarted the session, now its better. As for the above, here is what I can do:

[image]

So Is it still a usb dongle issue?

arch1mede, 2021-04-25:
I personally, am unable to repo this using my usb2hdmi dongle. I will admit I do have some issues with FF but switching to chrome resolved them. My usb device https://www.amazon.com/gp/product/B08FG54QPH

[image]

strausmann, 2021-04-25:
Good Day, this Is my usb dongle.

https://www.amazon.de/dp/B08BKNJP7N?ref=ppx_pop_mob_ap_share

arch1mede, 2021-04-27:
@strausmann Can you get on Discord so we can see if there is something else causing this? Otherwise id have to agree with the dev that its the usb dongle you got that is causing the issue.

strausmann, 2021-04-28:
A test with a CSI HDMI converter will show if the output of Chocolatey or the USB dongle is responsible for the faulty output. I will update the ticket once the converter has arrived and been tested.

```powershell
# output all the colour combinations for text/background
# http://stackoverflow.com/a/41954792/10245
$colors = [enum]::GetValues([System.ConsoleColor])
Foreach ($bgcolor in $colors){
    Foreach ($fgcolor in $colors) { Write-Host "$fgcolor|"  -ForegroundColor $fgcolor -BackgroundColor $bgcolor -NoNewLine }
    Write-Host " on $bgcolor"
}
```

closed 2021-04-28 by strausmann