pikvm/pikvm #287: I connected to a HDMI switch of aimos, but I couldn't find ttyusb in the /dev file

binbinshare, 2021-04-29:
I connect a USB-A to Micro USB cable from the Raspberry Pi4B to the management port on the side of the aimos switch.But  I couldn't find ttyusb in the /dev file,need any operation？

mdevaev, 2021-04-29:
AIMOS is not supported by the pikvm driver. We don't have a description of the protocol to do external management. Probably, AIMOS is not capable of this at all.

binbinshare, 2021-04-29:
[image]
This page has been written to test aimos. In addition, I tried to connect the mouse and keyboard directly to raspberry 4b, and there was no ttyusb file.

mdevaev, 2021-04-29:
This page does not say that there is a channel switching driver for aimos.

binbinshare, 2021-04-29:
Raspberry pi 4B keyboard and mouse, dev folder does not appear ttyusb file, normal？On raspberry pi 4b, a USB keyboard and aimos are connected. Enter dmesg, and the following message will be displayed：
[image]

mdevaev, 2021-04-29:
I'm confused. Is this log on the pikvm when connecting the AIMOS control port? If so, then everything is clear: the device is not recognized by the kernel. If AIMOS works the same way as some other switches I've seen, then it's actually not USB (although it uses this connector), but UART. You can't manage it from Linux.

binbinshare, 2021-04-29:
[image]
AIMOS switch port is usb,this log on the pikvm is  connecting a USB-A to Micro USB cable from the Raspberry Pi4B to the management port on the side of the aimos switch.
[image]

mdevaev, 2021-04-29:
Why do you think this is USB? Judging by what I see on the board, a button is connected to the lines on the "USB". It's definitely not USB.

Do you have this button? Disassemble it please:
[image]

GrafDe, 2021-05-03:
I have the same device. This is not a usb device. it's just a micro-USB connector in which two wires lead to the button. For this device to switch (between hdmi ports), it must be connected to a relay that will close these two wires like this button (as button in pic).

mdevaev, 2021-06-03:
Answered I guess

closed 2021-06-03 by mdevaev