pikvm/pikvm #289: macOS 11.3 Recovery Mode stuck at "no mouse" screen

KarimGeiger, 2021-04-30:
**Describe the bug**
With the latest version of macOS (11.3), the recovery mode cannot be used using PiKVM without an external mouse connected (on a Mac mini). When booting into recovery mode, the following screen appears, indicating that no mouse could be found.
[image]

This screen also appeared in the versions before, however, it could be dismissed by clicking the mouse/pressing a key. Now, there is no (known) way to work around this screen. Interestingly, the mouse cursor through PiKVM is begin shown and can be moved (i can even click the menu options on the top of the screen.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect a Mac with macOS 11.3 to PiKVM (I'm using an M1 model) without a mouse attached
2. Boot into recovery mode (Command + R on Intel, hold power button on M1)
3. Try to get past the "no mouse" screen

**Expected behavior**
The "no mouse" screen should either not show or should be dismissable.

**Screenshots**
see above

**Desktop (please complete the following information):**
 - OS: mac
 - Browser firefox
 - Version 88
 - VNC client (if used): no

**Pi-KVM info:**
  - Raspberry Pi board version: RPi 4
  - Pi-KVM platform: kvmd-platform-v2-hdmi-rpi4 2.59-1
  - Video capture type: CSI bridge
  - KVMD version: kvmd 2.59-1
  - uStreamer version: ustreamer 3.27-1
  - Linux kernel:  5.10.32-2-ARCH #1

mdevaev, 2021-04-30:
Hi. Do you use Arduino Hid?

KarimGeiger, 2021-04-30:
Hi @mdevaev, thanks for your quick reply! No, I'm not using Arduino for HID. Just the split USB cable on the Raspberry Pi.

mdevaev, 2021-04-30:
Try relative mouse mode: https://github.com/pikvm/pikvm/blob/master/pages/mouse.md

KarimGeiger, 2021-04-30:
Thanks for the tip. I just tried it and it seems to make no difference. I can still move the mouse on the Mac but the screen won't go away. I tried both modes (with horizontal_wheel on and off) and rebooted both devices, but no luck 😞

mdevaev, 2021-04-30:
Mac has some kind of generic mouse problem. I'm working with earlier macs and arduino hid (similar issue), but it looks like there's something very unusual there. Now I'm waiting for the USB protocol analyzer to figure this out.

KarimGeiger, 2021-04-30:
Thank you for looking into it. If I can be of any help, please let me know.

KarimGeiger, 2021-05-02:
Just had an idea: Since the mouse input seems to be working quite fine, I'm assuming that "no mouse" screen is an artificial lock put in place by Apple because they didn't recognize the USB device as a mouse. Therefore, would it be possible to just simulate a virtual mouse device that's connected via USB? (Similar to how you're simulating a USB stick?) I don't think it actually needs to do anything, I'm assuming it just needs to be there so macOS is happy.

Sorry if that makes no sense, though. I'm really not a hardware guy, otherwise I'd try verifying my idea myself.

mdevaev, 2021-05-16:
We are already emulating a USB mouse. This is some kind of bug on the Apple side, now we are figuring out how to solve it.

mdevaev, 2021-07-10:
I think I found out what the problem is. The fix for Arduino HID is already ready and will be available in the release soon, testing is needed for OTG and M1.

mdevaev, 2021-07-10:
Fixed. See details here: https://www.reddit.com/r/pikvm/comments/ohmx8j/kvmd_38_the_epic_apple_fix/

closed 2021-07-10 by mdevaev

KarimGeiger, 2021-07-13:
Thank you very much for fixing this! Really appreciated!