pikvm/pikvm #290: Video stream and KB+mouse works only in OS session

al-pankov, 2021-04-30:
**Describe the bug**
Video stream and KB+mouse works only in OS session. No stream and KB+mouse during BIOS post and so on until OS booted.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'KVM-mode' or connect using VNC
2. Reboot the PC
3. See error

**Expected behavior**
Previous version KVMd 2.8 and Ustreamer 2.2 works as expected on the same hardware.

**Screenshots**
No screenshots.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 90
 - VNC client: TigerVNC 1.11.0

**Pi-KVM info:**
  - Raspberry Pi board version: RPi 4
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 2.59-1
  - uStreamer version: 3.27-1
  - Linux kernel: Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux

**Additional context**
I've tested this error with desktop PC with discrete video card Nvidia GT730 and Lenovo laptop with integrated video card Intel UHD.

mdevaev, 2021-04-30:
Hello. See this: https://github.com/pikvm/pikvm/blob/master/pages/edid.md

mdevaev, 2021-04-30:
In the new versions, this logic has not changed, perhaps this is really an EDID problem.

closed 2021-05-07 by al-pankov

wcck, 2021-12-14:
Hi @al-pankov, @mdevaev 

How to enable HDMI on my Lenovo laptop when reboot ?
This NB is my remote system.
I want to connect it to operate BIOS's feature.
However BIOS didn't show any screen on reboot.

**Describe the bug**
No signal on reboot but Web UI will show screen until enter OS.

**Pi-KVM info:**
- Raspberry Pi board version: RPi 4
- Pi-KVM platform: v2-hdmi
- Video capture type: CSI bridge
- KVMD version: 3.48-1
- uStreamer version: 4.9-1
- Linux kernel: Linux pikvm 5.10.52-13-ARCH #1 SMP Sat Nov 20 09:40:57 MSK 2021 armv7l GNU/Linux

**Lenovo info. :**
- Model : ThinkPad 2 E14
- CPU : AMD Ryzen 5 4500U
- Graphics : AMD Radeon(TM) Graphics

**Host**
- OS: Windows 10
- Browser: Chrome
- Browser's Version: 96.0.4664.45

Thanks

wcck, 2021-12-14:
[Update]
I tried https://docs.pikvm.org/webrtc/#custom-janus-config and #78 .
However still didn't show any screen on UEFI.

al-pankov, 2021-12-14:
Laptop BIOS - Config - Display - Boot Display Device: HDMI

wcck, 2021-12-14:
Hi @al-pankov 
I tried success.
Thank you very much!